f4exb/sdrangel #1181: Crash during start with LimeSDR connected

arnobert, 2022-03-20:
* Compiled SDRAngel 6.20.1-14 from source on Arch
* Starts fine without LimeSDR connected
* With LimeSDR connected, I get a SIGSEGV
* LimeSDR version 20.10.0, GUI detects the board, GQRX works as well

srcejon, 2022-03-23:
It's working for me on Ubuntu.

Perhaps try running in gdb:

    gdb /path/to/sdrangel
    run

When it crashes:

    info stack

And then copy & paste the output to show where the crash is.

arnobert, 2022-03-23:
Hi,
good idea, here is the output:

<pre>Thread 1 &quot;sdrangel&quot; received signal SIGSEGV, Segmentation fault.
<font color="#0000AA">0x00007ffff48e7646</font> in <font color="#AA5500">libusb_free_transfer</font> () from <font color="#00AA00">/usr/lib/libusb-1.0.so.0</font>
(gdb) info stack
#0  <font color="#0000AA">0x00007ffff48e7646</font> in <font color="#AA5500">libusb_free_transfer</font> () at <font color="#00AA00">/usr/lib/libusb-1.0.so.0</font>
#1  <font color="#0000AA">0x00007ffff6875902</font> in  () at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#2  <font color="#0000AA">0x00007ffff68759a7</font> in  () at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#3  <font color="#0000AA">0x00007ffff6831d09</font> in <font color="#AA5500">lime::ConnectionRegistry::freeConnection(lime::IConnection*)</font> () at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#4  <font color="#0000AA">0x00007ffff6862c65</font> in <font color="#AA5500">lime::LMS7_Device::~LMS7_Device()</font> ()
    at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#5  <font color="#0000AA">0x00007ffff6868ca8</font> in  () at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#6  <font color="#0000AA">0x00007ffff68545f3</font> in <font color="#AA5500">LMS_Close</font> () at <font color="#00AA00">/usr/lib/libLimeSuite.so.20.10-1</font>
#7  <font color="#0000AA">0x00007fff09792a81</font> in <font color="#AA5500">DeviceLimeSDRParams::close()</font> ()
    at <font color="#00AA00">/usr/local/lib/sdrangel/liblimesdrdevice.so</font>
#8  <font color="#0000AA">0x00007fff0978c1ea</font> in <font color="#AA5500">DeviceLimeSDR::enumOriginDevices(QString const&amp;, QList&lt;PluginInterface::OriginDevice&gt;&amp;)</font> ()
    at <font color="#00AA00">/usr/local/lib/sdrangel/liblimesdrdevice.so</font>
#9  <font color="#0000AA">0x00007fff097b6f3f</font> in <font color="#AA5500">LimeSDRInputPlugin::enumOriginDevices(QStringList&amp;, QList&lt;PluginInterface::OriginDevice&gt;&amp;)</font> ()
    at <font color="#00AA00">/usr/local/lib/sdrangel/plugins/libinputlimesdr.so</font>
#10 <font color="#0000AA">0x00007ffff74bbbac</font> in <font color="#AA5500">DeviceEnumerator::enumerateRxDevices(PluginManager*)</font>
    () at <font color="#00AA00">/usr/local/lib/sdrangel/libsdrbase.so</font>
#11 <font color="#0000AA">0x00007ffff751c11e</font> in <font color="#AA5500">PluginManager::loadPluginsFinal()</font> ()
    at <font color="#00AA00">/usr/local/lib/sdrangel/libsdrbase.so</font>
#12 <font color="#0000AA">0x00007ffff7d47c94</font> in <font color="#AA5500">MainWindow::MainWindow(qtwebapp::LoggerWithFile*, MainParser const&amp;, QWidget*)</font> () at <font color="#00AA00">/usr/local/lib/sdrangel/libsdrgui.so</font>
--Type &lt;RET&gt; for more, q to quit, c to continue without paging--
#13 <font color="#0000AA">0x0000555555556de2</font> in <font color="#AA5500">runQtApplication(int, char**, qtwebapp::LoggerWithFile*)</font> ()
#14 <font color="#0000AA">0x0000555555556427</font> in <font color="#AA5500">main</font> ()
</pre>