f4exb/sdrangel #1185: QObject::connect: No such slot

srcejon, 2022-03-23:
It seems some of the recent API changes are resulting in slots not being connected.

Add any demod (E.g. AM Demod) and in the log, we get:

2022-03-23 02:44:06.420 (W) QObject::connect: No such slot ChannelAPI::networkManagerFinished(QNetworkReply*)
2022-03-23 02:44:06.420 (W) QObject::connect:  (receiver name: 'AMDemod')
2022-03-23 02:44:06.420 (W) QObject::connect: No such slot ChannelAPI::handleChannelMessages()
2022-03-23 02:44:06.420 (W) QObject::connect:  (sender name:   'MessageQueue')
2022-03-23 02:44:06.420 (W) QObject::connect:  (receiver name: 'AMDemod')

This may be because the demod classes (E.g. AMDemod) do not include the Q_OBJECT macro - or the c++ syntax needs to be used for connect instead of SLOT() macro.

f4exb, 2022-03-23:
I tend to think it is the second suggestion that would fix this. This does ring a bell from a similar situation already encountered.

srcejon, 2022-03-23:
Yeah - c++ syntax is also nicer as it's checked at compile time, rather than runtime.

commit 4111c967 (2022-03-23): Fixed some signal/slot connects. Removed unnecessary m_channelMessageQueue in ChannelAPI. Fixes #1185
closed 2022-03-23 by f4exb