f4exb/sdrangel #1188: Satellite tracks do not line up with the satellite icons

srs4511351, 2022-03-27:
I have a SDRplay RSP1A with API 3.07 running on a Raspberry Pi 4 B. I installed SDRangel on Debian Bullseye and Bookworm.

I have been enjoying receiving satellites recently.

I tried versions 6.19.1-18, 6.19.1-34 and 6.20.1-29. Satellite tracks do not line up with the satellite icons on the map. They are shifted somewhat.

The icon positions are correct as observed by acquisitions in the tracker list and signals received.
Updating the satellite data did not help.

Older versions before 3D maps were added, work correctly.
Using a .conf file from a working system did not help, but the configuration worked otherwise.

When using the replay feature in Satellite Tracker, with send time to map checked, the icons are correctly positioned on the ground tracks.

Is there a setting that I need to change to correct this?

----Steve

choupette-change, 2022-03-28:
Hello, same observation, the satellite tracks are no longer displayed (since the switch to summer time in Europe???)

srcejon, 2022-03-28:
Can you post a picture?

I see:

[image]

Which looks ok.

choupette-change, 2022-03-28:
ok in 2 D map but nothing in 3D map since the switch to summer time in Europe???
[image]

choupette-change, 2022-03-29:
It is ok for me

srs4511351, 2022-03-29:
What changed to make it OK you you? Did you change to UTC?
Your screen shot was not OK.

-----Steve

srcejon, 2022-03-29:
It should be fixed if you git update and rebuild.

choupette-change, 2022-03-29:
[image]

srs4511351, 2022-03-30:
I uninstalled, recompiled with the new version 6.20.1-37, and the satellite tracks do line up with the icons.
Thanks! Good work!

----Steve

f4exb, 2022-04-03:
Fix released in v6.20.2 hence can be closed

closed 2022-04-03 by f4exb