f4exb/sdrangel #1192: dot in a file

choupette-change, 2022-03-30:
Why is a dot added at the beginning of the file name? This file is no longer recognized unless you modify the name of this file by hand (remove the dot). Is-it a bug ??
[image]
[image]
[image]

f4exb, 2022-03-30:
Give he file a root name. Please read the documentation before posting an issue: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/filesink

choupette-change, 2022-03-30:
Oups I read the documentation before posting this bug but my English is light I did not see or understand the information Sorry

closed 2022-06-05 by github-actions[bot]