f4exb/sdrangel #1193: Revamping of the SDRangel main UI

f4exb, 2022-03-30:
This is the main point of the corresponding project: https://github.com/f4exb/sdrangel/projects/6

This ticket deals more specifically with the main UI organization and is the starting point of all dependent re-organizations. It further details the main UI organization and different components. 

As mentioned it revolves around the concept of workspaces where any kind of widgets can be placed together corresponding to their functional affinity rather than technical. In order to implement this kind of workspace in the Qt world the MDI area seems to fit best (QMdiArea). Then in order to be able to move a workspace to another screen in multi screen configuration the docking area concept becomes handy therefore this MDI area will be attached to an underlying QDockWidget. The main window will not have a central widget and each new workspace will be added to a dock widget area.

In the MDI area the devices, channels and main spectra can be placed and therefore the UI must derive from the QMdiSubWindow. By default a QMdiSubWindow has a non customizable top bar which is not really convenient therefore it should be customized by first removing the frame with the `Qt.FramelessWindowHint` window flag.

The `QMdiSubWindow` will be specialized into the device, channel or spectrum flavors. For the channel a new kind of RollupWidget without title bar will be designed (RollupContent for example) because the top bar will be handled in the containing window.

The present presets will be kept as "deviceset presets" and the feature set presets will be kept also so that the complete configuration can be assembled from pieces. However the global configuration may be saved in a full configuration preset to preserve the arrangements of all workspaces. As a matter of fact the assembly of the whole configuration is more complex and should be preserved across program starts.

Here is a Python/PyQt5 example to have a very basic idea and see how things may articulate at the top level: https://gist.github.com/f4exb/dfbfa0aa4397240d1d55110241e5c6c9

f4exb, 2022-04-03:
This is a preview of what it will look like. Only the features for now:
[image]

f4exb, 2022-04-09:
Progressing with the devices... Now able to show two running devices in the same workspace:
[image]

f4exb, 2022-04-13:
v7 branch is now online: https://github.com/f4exb/sdrangel/tree/v7 It works mostly but still need a few bug fixes, improvements and API updates.

f4exb, 2022-04-15:
Not completely there yet for a release candidate but I can build a fully functional dual receiver for Amateur VHF (Airspy HF) and UHF (Remote) all on the same display (workspace):
[image]

f4exb, 2022-04-19:
First alpha release is here: https://github.com/f4exb/sdrangel/releases/tag/v7.0.0-alpha.1

srcejon, 2022-04-20:
Excellent job :) This opens the door to a lot of interesting and powerful configurations! I also like that the frequency for each channel is now always displayed.

f4exb, 2022-05-02:
Deployed in v7.0.0-alpha.3
This concludes the UI features part.

closed 2022-05-02 by f4exb