f4exb/sdrangel #1195: No sound or waterfall with HF Discovery,M1 Mac

A2Sumie, 2022-04-03:
# SDRangel Issue Template
Please let us know how you use the software, and what you enjoy, or don't enjoy the most.  
We are interested to find out how other people make use of this software.

#### From the Project Wiki:  
   *I ask you (as a User or Collaborator) to take the role of first level support.   
   It means you have to help me and not the opposite: you have to help me  
   understand the problem or what you think is a problem*

### Subject or type of issue
* Describe your issue here.
No activity after clicking the play button. Sounds not coming and waterfall remains dark. HFD's blue LED indicator lits.
* Is this a runtime or compilation issue?
Seems not. 

### SDR
* We always want to first rule out that there is no issue with your hardware
* Which SDR are you using?
Airspy HF Discovery.
* Which version of Driver do you have loaded for the SDR?
Not sure actually since I'm not sure where to check. A recent driver for sure.
* If applicable, Can you prove SDR functionality with a loopback test?
Works fine with Sdrplusplus, Gqrx and CubicSDR on the same Mac.
* Have you tried power cycling the SDR or the machine connecting to the SDR? Sometimes a reset is all you need.
Rebooted my mac and reconnected. Won't help.
 * Perhaps this is the issue, that the issue causes you to need to reset.

### Development environment
* Compilation instructions are given in terms of Ubuntu 21.04, have you tried to recreate the issue using this environment?
Nope. Specific to this system I suppose.
* CPU and RAM of your machine if applicable
M1 max, 64 gigabytes.

### Actual behavior
* Tell us what happens.
Acts like the screenshot below. Sound and Waterfall won't come.
* Be specific here, It is important that we be able to recreate the issue
* Please list any relevant information leading up to the issue
* 
[image]

### Expected behavior
* Tell us what should happen
Proper operation or at least an error or some logs. Yet nothing happens.No logs are generated.
* Helps us ensure that you understand what you are trying to do so that we can rule out any user error.
Install using Macports, Choosing HFD from device list and clicking the purple play button.

### Steps to reproduce
* Tell us how to reproduce this problem. Include screenshots, where applicable
Use as normal.
* If radio related, have you tried to reproduce with a different radio, if possible?
* Please list Center Frequency, bandwidth and any other applicable RF settings that are pertinent to the issue.
1000k,1100k,3100k in HF mode, 6100k in VHF mode, nothing just happen.

### Log Files
* Please read any pertinant log files!
* Attach any relevant information you find here.

srcejon, 2022-04-03:
Do you have a USB 2 hub you can try it in? There is a problem on some versions of Linux where it only works on USB 2 hubs, not USB 3.

A2Sumie, 2022-04-04:
> Do you have a USB 2 hub you can try it in? There is a problem on some versions of Linux where it only works on USB 2 hubs, not USB 3.

Tried and nothing changed.
Did get a crash after several refreshes.

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               sdrangel [57210]
Path:                  /opt/local/bin/sdrangel
Identifier:            sdrangel
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [9600]
Responsible:           iTerm2 [9593]
User ID:               501

Date/Time:             2022-04-04 22:31:06.9699 +0900
OS Version:            macOS 12.3 (21E230)
Report Version:        12
Anonymous UUID:        9D865DC0-7CAC-EA2F-AB20-20E6CD4EF702

Sleep/Wake UUID:       F8798C1F-351A-438F-AEEE-E974BD358861

Time Awake Since Boot: 37000 seconds
Time Since Wake:       936 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x142735840000060 vs 0x735840000060)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0142735840000060 -> 0x0000735840000060 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0142735840000060
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [57210]

VM Region Info: 0x735840000060 is not in any region.  Bytes after previous region: 21269214920801  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)
--->  
      UNUSED SPACE AT END

Error Formulating Crash Report:
PC register does not match crashing frame (0x142735840000060 vs 0x735840000060)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x735840000060 ???
1   libusb-1.0.0.dylib            	       0x10140b1ac libusb_free_transfer + 64
2   libairspyhf.1.6.8.dylib       	       0x117b01b6c airspyhf_close + 172
3   libinputairspyhf.dylib        	       0x121e44134 AirspyHFInput::closeDevice() + 36
4   libinputairspyhf.dylib        	       0x121e44024 AirspyHFInput::~AirspyHFInput() + 140
5   libinputairspyhf.dylib        	       0x121e441ac AirspyHFInput::~AirspyHFInput() + 12
6   libsdrgui.dylib               	       0x100dcb8f8 MainWindow::sampleSourceChanged(int, int) + 172
7   libsdrgui.dylib               	       0x100dc6e20 MainWindow::samplingDeviceChanged(int, int, int) + 48
8   QtCore                        	       0x102f595ac 0x102e18000 + 1316268
9   libsdrgui.dylib               	       0x100db8a80 SamplingDevicesDock::deviceChanged(int, int, int) + 76
10  QtCore                        	       0x102f59528 0x102e18000 + 1316136
11  QtWidgets                     	       0x1026e6668 QAbstractButton::clicked(bool) + 48
12  QtWidgets                     	       0x1026e65b0 0x102648000 + 648624
13  QtWidgets                     	       0x1026e648c 0x102648000 + 648332
14  QtWidgets                     	       0x1026e6f60 QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 112
15  QtWidgets                     	       0x10267ba58 QWidget::event(QEvent*) + 108
16  QtWidgets                     	       0x10274f5d8 QPushButton::event(QEvent*) + 152
[... 519 lines cut ...]
  ],
  "experiments" : [

  ]
},
  "reportNotes" : [
  "PC register does not match crashing frame (0x142735840000060 vs 0x735840000060)"
]
}

Model: MacBookPro18,2, BootROM 7459.101.2, proc 10:8:2 processors, 64 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Feb  8 2022 01:51:44 version 20.90.39.0.8.7.114 FWID 01-510a2aee
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB2.0 HUB
USB Device: AIRSPY HF+
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.`

closed 2022-06-09 by github-actions[bot]