f4exb/sdrangel #1199: Rigctl control feature is not compatible with WSJT-X

majkrzak, 2022-04-05:
When setting up "Hamlib NET rigctl" radio rig controll in WSJT-X, "Test CAT" fails as commands are unsupported by `RigCtlServerPlugin`. 

Transmission dump of failed comunication:
```
\chk_vfo
RPRT -4
RPRT -4
\dump_state
RPRT -4
RPRT -4
```

Transmission from GQRX which works well:
```
\chk_vfo
RPRT 1
\dump_state
0
2
1
0.000000 10000000000.000000 0x2ef -1 -1 0x1 0x0
0 0 0 0 0 0 0
0 0 0 0 0 0 0
0xef 1
0xef 0
0 0
0x82 500
0x82 200
0x82 2000
0x221 10000
0x221 5000
0x221 20000
0x0c 2700
0x0c 1400
0x0c 3900
0x40 160000
0x40 120000
0x40 200000
0 0
0
0
0
0
0
0
0
0
0x40000020
0x20
0
0
v
VFOA
m
USB
2700
f
1000000
F 1000055.000000
RPRT 0
f
1000055
F 1000000.000000
RPRT 0
v
VFOA
s
0
VFOA
m
USB
2700
v
VFOA
s
0
VFOA
f
1000000
m
USB
2700
v
VFOA
s
0
VFOA
f
1000000
m
USB
2700
v
VFOA
s
0
VFOA
f
1000000
m
USB
2700
v
VFOA
s
0
VFOA
f
1000000
m
USB
2700
```

I hadn't find any proper documentation of `\dump_state` except https://github.com/gqrx-sdr/gqrx/issues/329 and https://github.com/Hamlib/Hamlib/blob/8fbda12ce2f435c4f1930d1fe642a10b9e12a3bb/tests/rigctl_parse.c#L4262

srcejon, 2022-04-30:
An alternative way to look at it, is what info does WSJT-X need from this command?

Does it need VFO support? As that isn't implemented either.

mdblack98, 2022-06-19:
Rather than trying to emulate rigctld GQRX should just emulate a standard rig like the TS-2000.
Hamlib can talk to any rig via TCP and it would be SOOOO much easier to just handle the TS-2000 protocol than the complex rigctld protocol.

mdblack98, 2022-06-19:
Then...other programs that don't know anything about rigctld can use virtual serial ports and rigctlcom to talk to GQRX too.

closed 2022-08-24 by github-actions[bot]

majkrzak, 2022-08-26:
I'm fine witch closing this as the rigctl require full rewrite, but closing it "as completed" is a bit overkill for me.

mdblack98, 2022-08-26:
Somebody turned on the no-issue-activity bot in github.  It auto closes things but probably leaves it open if it's tagged as an issue or such.

AsciiWolf, 2022-09-11:
@f4exb Any chance this issue could be reopened? Thanks!

evanbills, 2024-08-18:
This issue still persists.