f4exb/sdrangel #1202: B205mini not presenting itself when more then one SDR plugged in at a time

alphafox02, 2022-04-14:
Setup:
HackRF, RTLSDR, and Airspy plugged into laptop at the same time as a B205mini. Latest SDRAngel deb package on 20.04 as of today. 

Expected result:
All SDRs present themselves in the selection window. 

Actual result:
Every SDR expect the B205mini presents itself. 

Remedy:
Unplug every SDR except the B205mini and restart SDRAngel. 

Some output from Terminal as I unplug each SDR one by one and restart SDRAngel. This is after unplugging the RTLSDR, but the Airspy, HackRF, and B205mini remain plugged in.
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
2022-04-13 21:36:01.531 (D) DeviceUSRP::enumOriginDevices: exception:  LookupError: KeyError: key "name" not found in dict(NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE, NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE)

After the Airspy unplugged
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
2022-04-13 21:38:30.939 (D) DeviceUSRP::enumOriginDevices: exception:  LookupError: KeyError: key "name" not found in dict(NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE, NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE)

After the HackRF unplugged and only the B205mini remaining[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
2022-04-13 21:39:08.171 (D) DeviceUSRP::enumOriginDevices: found USRP device  "B205i[0:$1] 3102435"
2022-04-13 21:39:08.171 (D) DeviceUSRPParams::open: type=b200,name=B205i,serial=XXXXX,product=B205mini
[INFO] [B200] Detected Device: B205mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.

srcejon, 2022-04-16:
I would guess a driver conflict, rather than an SDRangel problem. Perhaps try with UHD 4 to see if that is any different.

alphafox02, 2022-04-16:
Doesn’t happen with any other application though (soapy based or native driver support), same machine and same radios plugged in. It’s probably not limited to just the b205mini itself, I suspect any multiple UHD setup may experience this.

srcejon, 2022-04-17:
I don't see a problem with UHD 4 on ubuntu 5.13.0-37 - with B210, RTLSDR, Airspy HF + SDRplay:

[image]

I don't have a HackRF though - so maybe try removing that and attaching other devices and seeing if it is the HackRF that is causing your issue.

(If you look at DeviceUSRP::enumOriginDevices - there isn't much SDRangel can do differently with regard to UHD)

alphafox02, 2022-04-17:
I’ll try, but I see you have soapy listed. I’m trying the 20.04 deb package. I was hoping to avoid compiling from source, but I’ll give that a try. There must be a way to resolve with uhd 3.15 though.

alphafox02, 2022-04-17:
Quick check;

b205mini and just the Airspy HF+ = only the Airspy seen

LimeSDRmini and the b205mini = both are selectable

I don’t think this is just a uhd 3.15 issue.

alphafox02, 2022-04-17:
Testing more, this time just an RTLSDR and the B205mini. I believe it's in the way SDRAngel is enumerating the devices as you can see here the RTLSDR is first seen before usrp, but then seen again by UHD and so the B205mini is not seen. Maybe something needs to "hold" the RTLSDR as in use or something so it's not then again enumerated during the usrp enumeration.

2022-04-17 15:08:48.076 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.rtlsdr
2022-04-17 15:08:48.092 (D) RTLSDRPlugin::enumSampleSources: enumerated RTL-SDR device #0
2022-04-17 15:08:48.092 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.remoteinput
2022-04-17 15:08:48.092 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.sdrplay
2022-04-17 15:08:48.095 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.sdrplayv3
2022-04-17 15:08:48.096 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.testsource
2022-04-17 15:08:48.096 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.usrp
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
2022-04-17 15:08:48.598 (D) DeviceUSRP::enumOriginDevices: exception:  LookupError: KeyError: key "name" not found in dict(NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE, NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE)

alphafox02, 2022-04-17:
This is from SDR++ with both the RTLSDR and B205mini plugged in at the same time, same PC and same UHD. As you can see both are seen, even by the UHD in this case. 

[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
[INFO] [b200_impl.cpp:386] [B200] Detected Device: B205mini
[INFO] [B200] Detected Device: B205mini

alphafox02, 2022-05-19:
Closing this as it’s fixed in v7. However, with a hackrf and b205mini plugged in you get a no name entry as a selection on top of hackrf and b205mini - I suspect because the uhd soapy is picking up the hackrf. I suggest keeping enumeration as is, but maybe just hide no name options from the drop down as uhd anything other then actual uhd equipment is terrible, like hackrf via uhd when the native hackrf driver is available is useless.

closed 2022-05-19 by alphafox02