f4exb/sdrangel #1203: Compiler error on v7

loewal, 2022-04-14:
# SDRangel Issue Template
Please let us know how you use the software, and what you enjoy, or don't enjoy the most.  
We are interested to find out how other people make use of this software.

#### From the Project Wiki:  
   *I ask you (as a User or Collaborator) to take the role of first level support.   
   It means you have to help me and not the opposite: you have to help me  
   understand the problem or what you think is a problem*

### Subject or type of issue

* Compilation issue:

 
[ 28%] Building CXX object sdrbench/CMakeFiles/sdrbench.dir/parserbench.cpp.o
/root/Downloads/sdrangel-7/sdrgui/mainwindow.cpp: In constructor ‘MainWindow::MainWindow(qtwebapp::LoggerWithFile*, const MainParser&, QWidget*)’:
/root/Downloads/sdrangel-7/sdrgui/mainwindow.cpp:325:5: error: ‘ui’ was not declared in this scope; did you mean ‘Ui’?
  325 |     ui->menuPreferences->removeAction(ui->action_LimeRFE);
      |     ^~

My environment:
PRETTY_NAME="Ubuntu 21.10"
NAME="Ubuntu"
VERSION_ID="21.10"
VERSION="21.10 (Impish Indri)"
VERSION_CODENAME=impish

#

f4exb, 2022-04-14:
You are not suppose to compile v7 unless you are a developer. Apparently you are not considering you cannot manage the error you receive.

loewal, 2022-04-14:
> You are not supposed to compile v7 unless you are a developer. Apparently you are not considering you cannot manage the error you receive.

Thanks for this rude answer.
Genius as a developer, autistic in relations.

kdutton1964, 2022-04-14:
I just agree he’s done a wonderful incredible job in providing what he has to the community! Just because you gotten your feelings hurt by his curt answer, doesn’t mean you need to lash out at him in a negative way. Hey how many others wouldn’t even give you the time of day. He at least looked at you email and responded to your communication.

f4exb, 2022-04-15:
I can understand you are eager to discover the latest nitty gritty. I am myself very excited about this however we're not quite there yet to publish even a release candidate. I these conditions what is on v7 is to be considered development/experimental and it is expected that you can solve some basic glitches like this yourself.

Anyway the cleanup is part of the latest commit so this can be considered as solved.

closed 2022-04-15 by f4exb

loewal, 2022-04-15:
Thank you Edouard.
With some adjustments I had v7 running on my iMac.
Have to get acquainted to rather new interface though.
Nice work!!
Thnx.

f4exb, 2022-04-15:
The new "quick start" is here: https://github.com/f4exb/sdrangel/wiki/Quick-start-v7 but detailed documentation is in progress.

loewal, 2022-04-15:
Nice Edouard!

To be successful I had to modify 2 things, one of which I asked for an explanation/solution…:

In the sdrangel-7/sdrgui/mainwindow.cpp, I had to remove the lines 322, 6 lines because of the compiler-error.
Just downloaded the new source and saw that you cleaned up some things. Now have to test this….

There is still this:
In sdrangel-7/fcdhid/fchidhid.h, I had to change "#include hidapi/hidapi.h" to ‘#include hidapi.h'
Used SED in my compilation-script to remove it.
No idea why this is not necessary on my Ubuntu-machine.

Thanks anyway for being so kind not to take my frustration too serious.

Regards,
Loe

f4exb, 2022-04-16:
I think that the error with hidapi comes from this issue: https://github.com/f4exb/sdrangel/issues/1160 so you would have the same on master (v6) branch also. It is just that it had to do a full recompilation because of changes in the makefiles. It looks like the system wide hidapi library is not installed the same way across systems. I had to make a custom cmake find module that may be does not work in all cases: https://github.com/f4exb/sdrangel/blob/master/cmake/Modules/FindHIDAPI.cmake

The Main Window UI is not built with Qt Creator anymore precisely to part from its rigid scheme of a central window with docking areas on the sides. Therefore the "ui" object is not there anymore.