f4exb/sdrangel #1204: No effect on "download flight information for selected aircraft" in ADS-B demod

daviderud, 2022-04-16:
As per title.

How to reproduce:
Register for a new free account at avionicstack.
Start the program
Setup ADS-B, put the API key in the setting dialog
Demodulate ADS-B data of planes, make sure it gets a callsign
Click the button.

In the log I see:
2022-04-16 13:31:47.721 (D) AviationStack::handleReply: data array is empty
2022-04-16 13:31:51.934 (D) ADSBDemodGUI::on_flightInfo_clicked - No flight number for selected aircraft

API counter is increasing in avionicstack so maybe it is just me unlucky, but so far I've never been able to get any data...

Manually searching same callsign on flightradar24, it exists (try for example "AFR47UN")

srcejon, 2022-04-16:
From the docs:

To be able to use this, a callsign for the highlighted aircraft must have been received. Also, the callsign must be mappable to a flight number, which is not always possible (this is typically the case for callsigns that end in two characters, as for these, some digits from the flight number will have been omitted).

Which looks like the case for you example. Fr24 has a better way of guessing the flight number.

daviderud, 2022-04-17:
I see. So far it seemed systematic for me, I'll try testing it out with more sample flights.

daviderud, 2022-05-04:
Today I tried with around 20 aircrafts - for all of them the callsign was received correctly - and no information was retrieved from avionicstack. Seem deterministic at least for me...

srcejon, 2022-05-04:
It appears aviationstack API is returning an "internal error" - I've contacted their support to see what the problem might be.

srcejon, 2022-05-09:
aviationstack seems to be working now.

Please try on a flight where the callsign is of the form: AAAnnn  - Where A are letters and n are digits. If there's letters at the end of the callsign, it probably wont work. Unfortunately that seems to be the case for many flights today.

daviderud, 2022-07-10:
Need more time to test

daviderud, 2022-08-26:
Confirmed working again, for some callsigns (majority AAAnnn but also sometime others).

closed 2022-08-26 by daviderud