f4exb/sdrangel #1206: No map ADS-B (all white).  Vers. 7.0.0 alpha1 Running on terminal.

500paolo, 2022-04-20:
No map on ADS-B demod (all white).
Vers. 7.0.0 alpha1 Running on terminal because on Mxlinux21.1 It's impossible to install by .deb (dependencies not satisfied).

2022-04-20 12:08:50.438 (C) ADSBDemodGUI::applyMapSettings - Failed to invoke createMap
2022-04-20 12:08:50.438 (D) ADSBDemodGUI::applyMapSettings - createMap returned a nullptr
2022-04-20 12:08:50.438 (D) ADSBDemodGUI::applyMapSettings - Couldn't find station

[sdrangel.txt](https://github.com/f4exb/sdrangel/files/8520795/sdrangel.txt)

srcejon, 2022-04-20:
In the log, you have:

2022-04-20 12:44:35.819 (W) qrc:/map/map.qml:5:1: module "QtPositioning" is not installed 
     import QtPositioning 5.12 
     ^
2022-04-20 12:44:35.819 (W) qrc:/map/map.qml:4:1: module "QtLocation" is not installed 
     import QtLocation 5.12 
     ^

So you need to install qtpositioning5 qtlocation5 packages.

500paolo, 2022-04-20:
Solved! Thank you very much [srcejon](https://github.com/srcejon).

closed 2022-04-20 by 500paolo