f4exb/sdrangel #1207: V7: Devices appear in wrong workspace

srcejon, 2022-04-20:
When reopening a configuration, devices can sometimes appear in the wrong workspace.

To recreate:

Workspace > New
Workspace > New
In W1 add Rx device > File Input
Switch to W0 tab
Exit SDRangel
Restart SDRangel
Exit SDRangel
Restart SDRangel
After the second restart, the FileInput device that was in W1 is now in W0, although its spectrum remains in W1.

f4exb, 2022-04-20:
Even simpler:
Workspace > New
Workspace > New
In W1 add Rx device > File Input
Exit SDRangel
Restart SDRangel
Exit SDRangel
Restart SDRangel

So it seems switching to W0 does not make the difference you just need to exit/restart twice.

f4exb, 2022-04-20:
Debug messages:
```
stop #1
2022-04-20 13:47:37.753 (D) MainWindow::saveConfiguration: sdrangel.samplesource.testsource device in workspace 1 spectrun in 1
start #2
2022-04-20 13:48:17.753 (D) MainWindow::loadConfiguration: add source sdrangel.samplesource.testsource in workspace 1 spectrum in 1
stop #2
2022-04-20 13:49:17.955 (D) MainWindow::saveConfiguration: sdrangel.samplesource.testsource device in workspace 0 spectrun in 1
```
So the error appears during the second stop

f4exb, 2022-04-20:
I think this is because the workspace index is saved in the device settings and thus the deserialization overwrites the index in the device GUI and this is what is saved at exit. I think the workspace index should be a member of the `DeviceGUI` class instead similarly to the `MainSpectrumGUI`.

f4exb, 2022-04-28:
Fixed in v7.0.0-alpha2

commit 007995df (2022-04-20): Massive UI revamping (v7): fixed device workspace index save/restore and removed useless geometry blobs. Fixes #1207
commit 249db03c (2022-04-20): Massive UI revamping (v7): fixed device workspace index save/restore and removed useless geometry blobs. Fixes #1207
closed 2022-04-28 by f4exb