f4exb/sdrangel #1209: V7: Additional sub window layouts beyond Tile & Cascade

srcejon, 2022-04-20:
I think it would be useful if there were some additional sub window layouts beyond the standard MDI Tile & Cascade, that were more specific to SDRangel (Obviously I appreciate you still have improvements planned!)

Tile, for example, doesn't appear to take in to account windows that should probably have fixed dimensions. For example, when Tiling the File Input, we get:

[image]

Something like this might look nicer:

[image]

(I haven't looked to see if this is actually a limitation with Tile - or if the window size hint isn't fixed vertically).

I think we could probably have a layout option similar to what I tried to do for the Features window, that stacks preferably fixed size windows, then splits up the remaining space between other windows. It could also possibly use knowledge of the type of window, to arrange them a little more logically (E.g. By device set/channel index etc).

I can take a look at implementing this, if helpful. Just assign any of my suggestions to me, if you think worth implementing.

f4exb, 2022-04-20:
I think the main issue is with the way fixed sized widgets or widgets that should have fixed size are handled. 

Firstly this has to be added to the constructor:
```c++
setSizePolicy(QSizePolicy::Fixed, QSizePolicy::Fixed);
```
Then the resize event should be handled and force a size adjustment like this:
```c++
void TestSourceGui::resizeEvent(QResizeEvent* size)
{
    adjustSize();
    size->accept();
}
```
Also the details of the `.ui` file in Qt Creator are not always set up correctly.

f4exb, 2022-04-20:
Same thing with channels and features ...

f4exb, 2022-04-21:
accidentally closed too early

srcejon, 2022-04-26:
I'm working on a MDI window layout that looks like the default layout for SDRangel V6. It's basically working, except I've noticed one small problem with sizing of the windows.

If I try to resize a window to the minimumSizeHint, it ends up 3 pixels higher. E.g. I effectively call:

    window->resize(window->minimumSizeHint())

Afterwards, window->height() == 3 + window->minimumSizeHint().height()

I think this may be something to do with this calculation:

void AMDemodGUI::resizeEvent(QResizeEvent* size)
{
    int minHeight = getRollupContents()->minimumHeight() + getAdditionalHeight();

Where:

    int getAdditionalHeight() const { return 25 + 22; }

If I reduce getAdditionalHeight by 3, it seems to work - but not sure where these values come from.

I see this used for FeatureGui and ChannelGUI - but although DeviceGUI has  getAdditionalHeight defined, it doesn't seem to be used.

srcejon, 2022-04-26:
Here's an example of a "V6" layouter

- Devices are stacked in numerical order top left.
- Fixed height features are stacked in order underneath devices and resized to match width of devices
- Channels are stacked in device then channel index order down the right and resized horizontally to match and then any available vertical space is shared being vertically expanding channels
- Remaining space in the middle is shared between spectrums and expanding features

[image]

I've added two buttons to the toolbar:

- One just applies this layout once when clicked
- The other is checkable and when checked, automatically applies the layout whenever windows are added/removed or the window is resized - which gives it V6 like behaviour

srcejon, 2022-04-26:
There seems to be a difference between the height of the top bar for different type of windows:

[image]

It appears the top object in the Device .ui files have a layout with margin of 2 pixels on each side.

[image]

However, most of the features and channels do not have a layout on the top object:

[image]

I measure the device height as 26px and feature/channel as 22px. Which makes sense from the above - 2 extra from topMargin and 2 extra from bottomMargin.

f4exb, 2022-04-26:
The bottom for the spectrum should be smaller now. Supposedly 10.

f4exb, 2022-04-28:
Implemented in v7.0.0-alpha2

closed 2022-04-28 by f4exb