f4exb/sdrangel #1213: V7: MDI Window Resizing

srcejon, 2022-04-22:
Currently, there are two size grips to allow resizing of MDI sub-windows - presumably because Qt::FramelessWindowHint is set and this disables the standard OS implemented resizing functionality.

However, a problem with this is that you can't resize windows by clicking and dragging their edges, which is the standard behaviour and something I find quite useful.

I had a quick play, and have written some code to support this, including changing the cursor:

[image]

Which also means we can remove the top size grip and use the standard graphics for the bottom grip. (Also it would be possible to get rid of the bottom grip as well if you wanted to - and perhaps just have all the buttons at the top to save that bit of space at the bottom - although I do like being able to see the channel freq! - could work for device / spectrum windows)

However, this does require a few pixels border/frame being added to the left and right sides. Are you OK with that? This border could possibly also be in a different colour, so that it was a little clearer which window is which when multiple windows are abutted.

f4exb, 2022-04-22:
I thought `Qt::FramelessWindowHint` was important to remove the ugly default Qt window top bar but apparently you have found a way to circumvent that. In this event resizing by the default grips could indeed be nicer. I am less sure about the border since this would make too many borders with the already existing rollup content border. Else the rollup content border has to be made more discrete in favor of the outer border.

srcejon, 2022-04-22:
It still uses Qt::FramelessWindowHint - I've just added some code to change the cursor and do the resizing when the mouse is clicked on the edge of the window, rather than rely on the OS's code.

The reason for adding the extra pixels, is that currently some of the widgets are right at the edge of the window, which means we can't detect button presses for resizing there. It still works in other areas where there aren't widgets - but probably not what users would expect.

If the buttons are at the bottom to avoid overcrowding, then OK. I just thought you might be trying to make use of the available space.

RE #1209 - for something like the AM demod, it might be worth fixing it only in the vertical direction, as it looks OK when resized horizontally, and allowing horizontal resizing may allow it to be sized such that it has the same width as other channels it's stacked on top of (so they all line up).

I will tidy up the code for this resizing, so you can try it and see what you want to do about the borders, if anything.

f4exb, 2022-04-22:
> for something like the AM demod, it might be worth fixing it only in the vertical direction

I think the trouble comes when you will use the tiling option then the widget becomes exaggeratedly wide. I have not found a way to control this it seems the "max width" setting is not operative.

f4exb, 2022-04-22:
Maybe I had not tried hard enough. It is indeed possible to have a maximum width although a bit tedious. However when tiling it will expand horizontally to the maximum and not limit itself to the minimum.

srcejon, 2022-04-22:
I'll have a look at doing a custom layout as per #1209 next week, if I can.

f4exb, 2022-04-25:
The cursor glitch was solved with PR #1215

f4exb, 2022-04-28:
Implemented in v7.0.0-alpha2

closed 2022-04-28 by f4exb