f4exb/sdrangel #1218: How can I reset SDRangel?

sefirot43, 2022-04-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

srcejon, 2022-04-27:
Do you mean you want to delete all settings?

On Windows, you can delete the registry entries under: HKEY_CURRENT_USER\SOFTWARE\f4exb\SDRangel

sefirot43, 2022-04-27:
Yes, I mean to delete all settings to start all over again.

closed 2022-07-03 by github-actions[bot]