f4exb/sdrangel #1219: Compile Problem

ku4b, 2022-04-28:
I am not sure what to make of this.  I compiled this program a number of times and it worked.  This time doing a make -j4
with sdrangel I see:

 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinksrv_autogen/mocs_compilation.cpp.o
/opt/build/sdrangel/plugins/channelrx/demodfreedv/freedvdemodsink.cpp: In member function ‘void FreeDVDemodSink::applyFreeDVMode(FreeDVDemodSettings::FreeDVMode)’:
/opt/build/sdrangel/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:436:13: error: ‘struct freedv_advanced’ has no member named ‘interleave_frames’
  436 |         adv.interleave_frames = 1;
      |             ^~~~~~~~~~~~~~~~~
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesink.cpp.o
make[2]: *** [pluginssrv/channelrx/demodfreedv/CMakeFiles/demodfreedvsrv.dir/build.make:102: pluginssrv/channelrx/demodfreedv/CMakeFiles/demodfreedvsrv.dir/freedvdemodsink.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:12987: pluginssrv/channelrx/demodfreedv/CMakeFiles/demodfreedvsrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkbaseband.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinksink.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksettings.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinksettings.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkwebapiadapter.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksender.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkfifo.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkmessages.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkplugin.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkwebapiadapter.cpp.o
[ 42%] Building CXX object pluginssrv/channelrx/sigmffilesink/CMakeFiles/sigmffilesinksrv.dir/sigmffilesinkplugin.cpp.o
[ 42%] Linking CXX shared library ../../../lib/pluginssrv/libremotesinksrv.so
[ 42%] Linking CXX shared library ../../../lib/pluginssrv/libsigmffilesinksrv.so
[ 42%] Built target remotesinksrv
[ 42%] Built target sigmffilesinksrv
make: *** [Makefile:152: all] Error 2

This is on Ubuntu 20.04.4 .  Any thoughts or pointers on what I missed?

Thanks,
Dave

f4exb, 2022-04-28:
Obviously there is a compatibility issue with the Codec2/FreeDV dependency. Make sure you follow the instructions in particular checking out the exact commit: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#codec2freedv

ku4b, 2022-04-28:
Thanks!  Will do.