f4exb/sdrangel #1220: V7: Changing device results in crash

srcejon, 2022-04-28:
To recreate:

- Add FileInput device
- Press Change Device button in device window
- Change to Test Source (or any other source)
- Crash

f4exb, 2022-04-29:
This shows how Qt is bad. This is a showstopper for v7.

f4exb, 2022-04-30:
OK I think I've found out... It seems to be related to the fact that the GUI is not destroyed before the new one is created. The line has been commented out because it was making the program crash. The reason is invisible if not in debug mode because of optimizations. In debug mode it shows this happens in the destructor of `SamplingDeviceDialog`. The dialog is created on the stack of the function with "this" (`DeviceGUI`) as the parent thus when this `DeviceGUI` is destroyed the function has already been exited and  `SamplingDeviceDialog` freed. But since `SamplingDeviceDialog` was attached to the widget the destructor of `QWidget` attempts to delete it a second time. 

Thus two possibilities. Either

  - Do not attach the dialog to the widget in the first place (let it have a nullptr parent)
  - Nullify the parent with `setParent(nullptr)` before emitting the signal that will trigger the device change and hence the deletion of the old one.

f4exb, 2022-04-30:
In addition it appears that the new GUI is created as if the old one was still holding the place and thus it is shifted somewhere else in the workspace. A fix saving and restoring the position works around this. If the size of the new window is too big the user can move it somewhere else or re-arrange the workspace (auto stack overcomes this).

f4exb, 2022-05-02:
Deployed in v7.0.0-alpha.3

commit fcb2906f (2022-04-30): Massive UI revamping (v7): fixed device change. Fixes #1220
closed 2022-05-02 by f4exb