f4exb/sdrangel #1222: Decommission SDRplay (v1) in favor of SDRPlayV3

f4exb, 2022-04-30:
Thus removing dependency on libmirisdr4 that wants to live its own life.

f4exb, 2022-04-30:
The legacy SDRplay v1 and its Chinese clones are well supported by V3.

f4exb, 2022-04-30:
After thorough checking it appears that with V3 and SDRplay RSP1 there are short discontinuities in the IQ flow so we should keep the libmirisdr4 v1.1.2 (v2.0.0 does not work) based version.

closed 2022-04-30 by f4exb

srcejon, 2022-04-30:
Is this something that needs fixing in the sdrangel v3 code or that might affect other radios?

(I don't have an RSP1 to test)

f4exb, 2022-05-01:
It is only with a Chinese clone of RSP1 (the blue "MSi.SDR"with a dolphin logo). The RSP2 works fine.

f4exb, 2022-05-07:
This one (not all of them have the logo):
[image]