f4exb/sdrangel #1223: Decommission VOR channel plugin in favor of the VOR  "single channel" plugin + VOR localizer feature

f4exb, 2022-04-30:
The title says it all. The latter is has better performance and fits the original design that requires that one channel should deal with only one flow.

srcejon, 2022-05-02:
The Navaids aren't as up to date in the single channel version, IIRC. That can be updated, of course.

f4exb, 2022-05-02:
The Navaids code is in the VOR localizer feature not in the VOR channel. Original code was split into the channel part for what concerns signal processing and the feature part for what concerns the post processing (map, navaids ...) and channel(s) control. I saw some commits applied to both the original (multi) channel VOR and the VOR localizer so I am not sure about what you refer to. Also it seems to me that the Navaid data is downloaded from a source on the Internet and that URLs are the same in the original VOR channel and the VOR localizer.

f4exb, 2022-05-02:
Deployed in v7.0.0-alpha.3

closed 2022-05-02 by f4exb

srcejon, 2022-05-03:
Looks like both were out of date (so they weren't downloading valid VOR databases - but would work if you'd previously downloaded them). I've just updated it to share the latest code with the ADS-B demod. I will add a PR in a bit, but it seems there are a couple of other issues with the SC version I'll look at first: It doesn't work with FileInput and there's an occasional crash when selecting VORs.