f4exb/sdrangel #1224: CW plugin

OK1UNL, 2022-04-30:
Hi friends,
is a possible make an CW plugin with setting CW shift, Bandwidth, Order and Filter model ( Blackman Harris4, Blackman, Hamming, Blackman Harris7  etc.... ) 
Thank you for participation and effort.
VY 73! Ladislav OK1UNL

f4exb, 2022-04-30:
Isn't the SSB demodulator enough for you? Essentially SSB and CW is the same kind of processing. What do you think is missing in the SSB demodulator?

OK1UNL, 2022-05-01:
You have right, but.... CW shift mean BFO pitch for ( conventional RX like Up or DWN converter) i prefere cca 700Hz.
So I am not pure CW operator. CW bandwidth cca 100Hz. For SSB you need cca 2k4 Hz or slightly narrow to 1k8 Hz. This mean for quick tune from SSB to CW sub-band or change mode complete setting these. CW mode plug is going to allow pre-setting SSB and CW demodulators.    
VY 73! TNX. ladislav OK1UNL

f4exb, 2022-05-01:
You can do all this with the SSB demodulator:

> CW shift mean BFO pitch for ( conventional RX like Up or DWN converter) i prefere cca 700Hz.

Move the center frequency -700 (USB) or +700 Hz (LSB) away from the frequency of the CW signal

> CW bandwidth cca 100Hz

You can control the bandwidth with the lower and upper frequency filter limits

> CW mode plug is going to allow pre-setting SSB and CW demodulators.

All these settings are saved in the presets. You can have one demodulator for SSB and one for CW and mute one or the other depending on your usage.

f4exb, 2022-05-01:
> Order and Filter model ( Blackman Harris4, Blackman, Hamming, Blackman Harris7 etc.... )

Indeed the filter is a FFT filter and you could possibly choose between different windows if this is useful anyway.

f4exb, 2022-05-02:
About 7 term Blackman Harris: https://dsp.stackexchange.com/questions/51095/seven-term-blackman-harris-window

OK1UNL, 2022-05-02:
> You can do all this with the SSB demodulator:
> 
> > CW shift mean BFO pitch for ( conventional RX like Up or DWN converter) i prefere cca 700Hz.
> 
> Move the center frequency -700 (USB) or +700 Hz (LSB) away from the frequency of the CW signal
> 
> > CW bandwidth cca 100Hz
> 
> You can control the bandwidth with the lower and upper frequency filter limits. As the slider moves by 100 Hz you can narrow it down to 100 Hz bandwidth. To have 700 Hz in the middle you can adjust the center frequency with a step as small as 1 Hz.
> 
> > CW mode plug is going to allow pre-setting SSB and CW demodulators.
> 
> All these settings are saved in the presets. You can have one demodulator for SSB and one for CW and mute one or the other depending on your usage. Or to be more handy one can imagine having a bank of filters so you can switch from one to another with the same SSB demodulator.

Thank you for explanation. I tried it and works well. So me have 2 instances Airspy Discovery and Airspy R2 both simultaneously on one PC.  Good results.

OK1UNL, 2022-05-02:
Thank you for patience and I am tried your suggests, with good results. 
Please you can close this topic.
VY 73! Ladislav OK1UNL

f4exb, 2022-05-02:
Some suggestions for the SSB demod are interesting or at least deserve checking:

  - Having a bank of filters to be able to switch quickly from one to another. The span must be taken into account also because it conditions the possible limits of the filter
  - See if selectable FFT windows brings something. This would be also part of the filter setting in the bank of filters.

f4exb, 2022-05-19:
Implemented in v7.0.1

closed 2022-05-19 by f4exb