f4exb/sdrangel #1225: Crash due to "illegal instruction" on Fedora/QubesOS

kord1e, 2022-05-01:
SDRangel v. 6.20.1, installed from repositories, crashes on QubesOS/ Fedora 34 VM when it is started. Nothing except this is printed to the console:

Illegal instruction (core dumped)

In dmesg this can be seen:

[  155.479745] traps: sdrangel[1790] trap invalid opcode ip:7b89f61578b2 sp:7ffcc8f98480 error:0 in libsdrbase.so[7b89f606a000+151000]

Similar bug has been filed at Arch Linux tracker in 2019:
https://bugs.archlinux.org/task/62675

Please let me know if there's any other information that would be useful.

srcejon, 2022-05-01:
What CPU do you have?

kord1e, 2022-05-01:
@srcejon 

CPU is Intel Xeon E3-1230 V2

lscpu reports the following flags:

fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht syscall nx rdtscp lm constant_tsc rep_good nopl cpuid tsc_known_freq pni pclmulqdq ssse3 cx16 pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c hypervisor 
                                 lahf_lm cpuid_fault pti ssbd ibrs ibpb stibp fsgsbase smep erms xsaveopt md_clear flush_l1d

f4exb, 2022-05-01:
Do not trust packages and compile the code yourself: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

kord1e, 2022-05-07:
Thanks, I compiled it myself and it works now.

But it seems to be somewhat common that distro packagers compile this software with improper settings. I think it would be useful if the build instructions were updated to address this, so that packagers would be aware of this issue. Or the build configuration could be updated to use default settings that work on any platform?

kord1e, 2022-05-20:
Looks like it's working on Fedora/Qubes now, apparently the packagers fixed it. I'm closing this for now.

closed 2022-05-20 by kord1e