f4exb/sdrangel #1226: hackref and sdrangel problem tx

ik5wwc, 2022-05-01:
Salutations !! i installed sdrangel on linux mint.can hackref one ..
but i have encountered problems in tx.
When I call the T1 window in NFM modulator, sdrangel closes.
in SSB modulator the TX window opens, but as MIC active, sdrangel closes. TX In AM it seems to work fine.
when i go to tx, obviously i deactivate rx.
is there a chance to solve this problem? Thanks Mauro

f4exb, 2022-05-01:
HackRF is a half duplex device so you cannot Rx and Tx at the same time but switching over is possible. It is easier if you use the PTT feature. It is also easier to demonstrate with the v7 GUI where you can see every piece on the same screen. This works perfectly:
[image]

ik5wwc, 2022-05-02:
Hi .. sorry I didn't explain my English and made in google ... RX in off, when I go to load the TX (e.g. T1) both in fm NFM modulator, DATV modulator, sdrangel closes and I am on the desktop . In AM modulator it's ok. SSB modolator ok, but when I activate the microphone, the sdrangel closes and I am in the desktop of the computer.
I use sdrangel also with windows 10 but with the soapysdr drive and everything works, but with the hackref drive, sdrangel closes and I am on the desktop. I tried to install soapysdr for linux but sdrangel does not see it ..
I thank you for your patience 73 de mauro.

Ciao.. scusami non mi sono spiegato il mio inglese e made in google...RX in off,quando vado a caricare il TX (es.T1) sia in fm NFM modulator,DATV modulator,sdrangel si chiude e mi trovo sul desktop. In AM modulator è ok. SSB modolator ok,ma quando attivo il microfono,sdrangel si chiude e mi trovo nel desktop del computer.
utilizzo sdrangel anche con windows 10 però con il drive soapysdr e funziona tutto,per con il drive hackref , sdrangel si chiude e mi trovo sul desktop.Ho provato a installare soapysdr per linux ma sdrangel non lo vede..
ti rigrazio della tua pazienza 73 de mauro.

closed 2022-07-07 by github-actions[bot]