f4exb/sdrangel #1228: Create Flatpak Distribution

n1zzo, 2022-05-02:
Hello,
sdrangel is my top favourite SDR software, however the only way to have a complete setup with all the plugins is to compile everything manually from source. as I cannot use the docker image because the GUI only run through X Forwarding.
It would be amazing to package a plugin-complete version of sdrangel into a Flatpak that could be run anywhere regardless of the OS native libraries.
Could something like that be of interest of the maintainers?

f4exb, 2022-05-02:
Sorry no flatpack, no snap or the like: https://github.com/f4exb/sdrangel/wiki#snap-and-flatpak

However if someone feels like creating a flatpack image as third party one can of course do so. As mentioned in the Wiki there are flatpack and snap folders but I believe the contributors have given up for a long time already so these are kept only as possible starting points.

n1zzo, 2022-05-02:
Thanks for the quick reply!
I will meditate on creating a Flatpak image for sdrangel and eventually open a PR.
I'm closing this issue now.

closed 2022-05-02 by n1zzo