f4exb/sdrangel #1233: Missing "feature/feature.h"

gvanem, 2022-05-04:
I have some problems compiling stuff that uses/needs the `class Feature`. E.g. while compiling the *Meta Object Code* for `plugins/feature/ais/aisgui.h`, there is this error from MSVC:
```
aisgui.h(53): error C2061: syntax error: identifier 'Feature' 
aisgui.h(87): error C2061: syntax error: identifier 'Feature' 
```

I found that a `#include "feature/feature.h"` is **not** present in that compile unit. So simply patching:
```diff
--- a/sdrgui/feature/featuregui.h 2022-05-04 11:10:58
+++ b/sdrgui/feature/featuregui.h 2022-05-04 12:51:39
@@ -21,6 +21,7 @@
 #include <QMdiSubWindow>
 #include <QMap>

+#include "feature/feature.h"
 #include "gui/framelesswindowresizer.h"
 #include "gui/rollupcontents.h"
 #include "export.h"
```
it now compiles. But I fail to understand how it builds on e.g. CI w/o such a patch.

f4exb, 2022-05-04:
I suggest to wait until merge of v7. In v7 there is no reference of `Feature` in `FeatureGUI`.

f4exb, 2022-05-04:
I cannot see any "Feature" in the source code of both featuregui.h and featuregui.cpp so I cannot see why it would possibly require to include feature/feature.h

f4exb, 2022-05-04:
OK I see... in fact your patch creates an unnecessary dependency in FeatureGUI instead in principle "feature/feature.h" should be included in aisgui.cpp and "Feature" made a forward declaration in aisgui.h. However this does not explain why my code compiles and yours not so there is yet another thing somewhere.

gvanem, 2022-05-05:
My repos seemed messed up. So I deleted everything and checked out again in a sub-tree:
```
md Git-latest-v7
git clone https://github.com/f4exb/sdrangel.git Git-latest-v7
git checkout v7
```

Correct? 

It works much better now, but there is a still a unresolved `Feature` in `plugins/feature/ais/aisgui.h`:
https://github.com/f4exb/sdrangel/blob/v7/plugins/feature/ais/aisgui.h#L53

And in other places, I still get other build-errors. Like for the *Meta Object Code* for 
`sdrgui/gui/devicesetpresetsdialog.h`:  `gui/devicesetpresetsdialog.h(65): error C2143: syntax error: missing ';' before '*'`

But I added patches for such errors and others:
```diff
--- a/plugins/channelrx/demodadsb/adsbdemodsinkworker.h 2022-05-05 11:34:05
+++ b/plugins/channelrx/demodadsb/adsbdemodsinkworker.h 2022-05-05 11:39:45
@@ -25,6 +25,7 @@
 #include "util/crc.h"
 #include "util/messagequeue.h"
 #include "adsbdemodstats.h"
+#include "adsbdemodsettings.h"

 class ADSBDemodSink;
 struct ADSBDemodSettings;

--- a/plugins/channelrx/localsink/localsinksink.h 2022-05-05 11:34:12
+++ b/plugins/channelrx/localsink/localsinksink.h 2022-05-05 11:39:45
@@ -22,6 +22,7 @@
 #include <QThread>

 #include "dsp/channelsamplesink.h"
+#include "dsp/samplesinkfifo.h"

 #include "localsinksettings.h"

--- a/plugins/channeltx/mod802.15.4/ieee_802_15_4_modsource.h 2022-05-05 11:34:13
+++ b/plugins/channeltx/mod802.15.4/ieee_802_15_4_modsource.h 2022-05-05 12:30:04
@@ -35,6 +35,7 @@
 #include "util/lfsr.h"
 #include "util/movingaverage.h"
 #include "util/message.h"
+#include "util/messagequeue.h"

 #include "ieee_802_15_4_modsettings.h"
--- a/plugins/feature/afc/afcworker.h 2022-05-05 11:34:13
+++ b/plugins/feature/afc/afcworker.h 2022-05-05 11:39:45
@@ -24,10 +24,12 @@

 #include "util/message.h"
 #include "util/messagequeue.h"
+#include "afc.h"

 #include "afcsettings.h"

 class WebAPIAdapterInterface;
+class DeviceAPI;
 class DeviceSet;
 class ChannelAPI;

--- a/sdrbase/util/messagequeue.h 2022-05-05 11:34:24
+++ b/sdrbase/util/messagequeue.h 2022-05-05 11:39:45
@@ -22,10 +22,9 @@
 #include <QObject>
 #include <QQueue>
 #include <QMutex>
+#include "util/message.h"
 #include "export.h"

-class Message;
-
 class SDRBASE_API MessageQueue : public QObject {
 	Q_OBJECT

--- a/sdrgui/feature/featuregui.h 2022-05-05 11:39:28
+++ b/sdrgui/feature/featuregui.h 2022-05-05 12:40:41
@@ -21,6 +21,7 @@
 #include <QMdiSubWindow>
 #include <QMap>

+#include "feature/feature.h"
 #include "gui/framelesswindowresizer.h"
 #include "gui/rollupcontents.h"
 #include "export.h"

--- a/sdrgui/gui/commandsdialog.h 2022-05-05 11:39:28
+++ b/sdrgui/gui/commandsdialog.h 2022-05-05 12:09:27
@@ -20,6 +20,7 @@
 #define SDRGUI_GUI_COMMANDSDIALOG_H_

 #include <QDialog>
+#include <QTreeWidgetItem>

 #include "export.h"

@@ -29,6 +30,7 @@
     class CommandsDialog;
 }

+class Command;
 class SDRGUI_API CommandsDialog : public QDialog
 {
     Q_OBJECT

--- a/sdrgui/gui/devicesetpresetsdialog.h 2022-05-05 11:39:28
+++ b/sdrgui/gui/devicesetpresetsdialog.h 2022-05-05 12:15:23
@@ -21,6 +21,7 @@

 #include <QDialog>
 #include <QList>
+#include <QTreeWidgetItem>

 #include "export.h"
```

I'm not sure which Qt version you're using. I use Qt 5.15.2. So the above could be due to my Qt being too old.
But I'm not able to build using Qt6.

srcejon, 2022-05-05:
Qt 5.15.2 works for me with MSVC.

What version of MSVC are you using? I haven't tried with 2022 yet.

gvanem, 2022-05-05:
MSVC-2019. From `cl /?`:  *Version 19.29.30142.1 for x86*.
This must be an issue with the horrible CMake generated stuff.

srcejon, 2022-05-05:
Are you building with ninja? (cmake -G ninja)

closed 2022-07-12 by github-actions[bot]