f4exb/sdrangel #1236: HackRF and SDRangel TX issues.

DJRaov, 2022-05-06:
OS: Arch Linux 
SDRAngel version: 6.20.2

Hello,
I have been struggling with getting HackRF work in SSB with SDRAngel. When trying to send ANY signal over to get modulated (be it a built-in carrier generator or microphone input) SDRAngel crashes with `free(): invalid pointer` error. This happens on every USB port, with every HRF firmware version. Even with a test sink.

Full log: `2022-05-06 18:23:13.921 (D) SSBMod::handleMessage: MsgConfigureSSBMod
2022-05-06 18:23:13.921 (D) SSBModBaseband::handleMessage: MsgConfigureSSBModBaseband
2022-05-06 18:23:13.958 (D) FFTFactory::releaseEngine: engineSequence: 1 FFT fwd size: 1024
free(): invalid pointer`

I have seen another thread similar to mine, but it did not help.
Anything I could try?

f4exb, 2022-05-17:
easily reproducible on current master.

f4exb, 2022-05-17:
Call to the pure virtual `feed` function of `BasebandSampleSink` triggers the destructor of `SpectrumVis` which is hardly understandable. However there is no need to manipulate the base class `BasebandSampleSink`. Replacing by direct call to `SpectrumVis` fixes the issue.

commit cc8b2e64 (2022-05-17): Fixed audio FIFO write overflows in modulators using audio input. Also fixes #1236 in SSB modulator
closed 2022-05-17 by f4exb