f4exb/sdrangel #1237: Problem in  ADS-B demodulator

mosta7il, 2022-05-08:
1. sampling rate problem:
        error message : Sample rate must be >= 2000000. Currently 64000.
       though I chose SR = 12 M
2. No airplanes appear
3. Is there any document which I can follow to make ADS-B works,
   Is there any restrictions on anntena connected to SDR
   
   Thanks in advance

srcejon, 2022-05-08:
> 1. sampling rate problem:
>    error message : Sample rate must be >= 2000000. Currently 64000.
>    though I chose SR = 12 M

This is referring to the baseband sample rate, which is set by the device sample rate and the decimation (Dec) setting. Which SDR are you using? If RTL SDR, set it's sample rate at 2,400,000 and Dec to 1. Perhaps attach a screenshot.

The SR setting in the ADS-B window is the channel sample rate - I'd recommend having that as 4.

> No airplanes appear

They wont until baseband sample rate is >= 2MSa/s 

> Is there any restrictions on antenna connected to SDR

Not really - just one that works well at 1090MHz. I use one of these LNAs as well which helps: https://store.uputronics.com/index.php?route=product/product&product_id=50

mosta7il, 2022-05-08:
> Which SDR are you using? If RTL SDR, set it's sample rate at 2,400,000 and Dec to 1.
Yes, I use RTL SDR
>  set it's sample rate at 2,400,000 and Dec to 1.
i set it as you say is say something i missed ?
[image]
this my used antenna 
[image]

srcejon, 2022-05-08:
Try turning device gain up to maximum. 

Set ADSB threshold to 5.

Under Preferences > My location set yor correct latitude and longitude. 

Try with antenna outside withonly a short amount of cable to SDR.

mistergitj, 2022-05-24:
I have found that even though I am in USA, I need to set my Longitude to a minus number to have home on the correct ADS-B Map.
Thanks and enjoy, John.

srcejon, 2022-05-24:
Yes, East and North are positive.

mistergitj, 2022-05-24:
Thanks, srcejon. I always thought that West was positive. Enjoy, John.

closed 2022-07-29 by github-actions[bot]