f4exb/sdrangel #1238: is there a way to record in sdrangel

mosta7il, 2022-05-08:
sorry ; but am new to this program and can't find where record button 
so is there a way to record one channel or multiple channel
thanks

srcejon, 2022-05-09:
Add a File Sink channel: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/filesink/readme.md

mosta7il, 2022-05-09:
i make a file and name it test.sdrig
[image]

but this message appear 
[image]

srcejon, 2022-05-09:
On Windows, Program Files is a protected directory, so you can't write to it (Unless program is run as Administrator)

Create the file in your User Documents directory or somewhere else.