f4exb/sdrangel #1241: Impossible to load maps with Ubuntu 22.04 LTS

choupette-change, 2022-05-09:
hello, impossible to load maps with Ubuntu 22.04 LTS, the latest version of Ubuntu with v7.0.0-alpha.3 and v6.20.3, ok with Windows
[image]

choupette-change, 2022-05-18:
Hello, am I the only one who has this problem ? Can you confirm that it works with Ubuntu 22 04 LTS. Best regards

loewal, 2022-05-18:
Tried to load the MAP feature under Linux.
System hangs!
Other 'features' are okay

loewal, 2022-05-18:
Tried the same on iMac and Macbook Pro. Maps are loaded fine.

srcejon, 2022-05-18:
This is probably the same problem as #1169 - in that 22.04 appears be using Qt 5.15.3

srcejon, 2022-06-15:
Workaround implemented in 7.3.2.

closed 2022-06-15 by srcejon