f4exb/sdrangel #1244: install not working

Alle1307, 2022-05-13:
I have tried to install SDRAngel in Ubuntu 21.10 and got errors compiling, Also the snap from the Ubuntu Software store is failing. I have upgraded to 22.04 but the same issues is coming back. It looks like a GUI problem. I have also installed QT5 and 6 but the same issue appears.
This is the error report in Dutch I hope it is readable

Consolidate compiler generated dependencies of target demodwfmsrv
[ 33%] Built target udpsinksrv
make[2]: *** Er is geen regel om doel '/usr/lib/x86_64-linux-gnu/libOpenGL.so' te maken, nodig voor 'lib/libsdrgui.so'.  Gestopt.
make[1]: *** [CMakeFiles/Makefile2:4547: sdrgui/CMakeFiles/sdrgui.dir/all] Fout 2
make[1]: *** Wachten op onvoltooide taken...
Consolidate compiler generated dependencies of target filesinksrv
[ 33%] Built target localsinksrv
[ 33%] Built target demodwfmsrv
[ 33%] Built target filesinksrv
make: *** [Makefile:156: all] Fout 2

Does somebody know what is failing?

f4exb, 2022-05-14:
So... "There is no rule to create target '/usr/lib/x86_64-linux-gnu/libOpenGL.so', needed for 'lib/libsdrgui.so" means you do not even have OpenGL installed.

f4exb, 2022-05-14:
Do `sudo apt install libopengl-dev`. I thought `libqt5opengl5-dev`would have OpenGL as a dependency but this is not the case apparently. I will add it to the list of dependencies in the wiki.

Alle1307, 2022-05-14:
Edouard that solved the issue indeed. 
73 Alle PE0AJF

closed 2022-05-14 by f4exb