f4exb/sdrangel #1246: NOAA (162.55 MHz) does not work on Beta.

mistergitj, 2022-05-15:
I have been trying the Beta version on W10 x64 using both an AirSpy and a NooElec R820T.  It looks like local FM works OK but I tried tuning NOAA (162.55 MHz), local in my area.  The signal is very weak and the NBFM and WBFM cannot decode it.  Both radios work on other SW, SDR#,HDSDR,cubicSDR with not problems. The signal is strong.  I also tried 6.20.3 with the same results. Thanks for any comments, John.

srcejon, 2022-05-15:
A screenshot might help

dziban303, 2022-05-16:
Works fine for me in both beta 2 and 3, coming through surprisingly well on a $15 RSP1 clone with a paperclip antenna. So it sounds like a settings issue on your end. Post some screenshots.

mistergitj, 2022-05-16:
Thanks to srcejon and dziban.  Please look at my screenshot.  Comments welcome.
Seems to be working better, particularity when I set the sample rate to 2.5 MSps.  Here is a screenshot:
[image]
Probably don't need any changes.  The audio is a little clearing on my other radios.  I am still learning your application. Thanks, John.

dziban303, 2022-05-16:
If you run into an issue again with an apparently clear signal not being
output to audio, on the NFM Demod panel, try hitting the Δ symbol next to
Sq on the right-hand side. You may need to play with the dials to its
right.

This program definitely has a learning curve, haha. That's half the fun!
I've been using SDRangel for some time now and there are still features I
haven't gotten to yet. Don't forget, if you hover over buttons and dials, a
popup will appear telling you what it is. And if you click the ? icon at
the top right of a pane it will open the relevant manual page.

Cheers

mistergitj, 2022-05-16:
dziban, thanks for your help.  I have now tried the Squelch (Δ) and it helps, but because of the weak signal, it is hard to understand.  It is also a function of the time of day.  I had failed to investigate the ?, so I did.  Thanks, John.

srcejon, 2022-05-17:
You have DC filtering enabled and your signal is at the centre frequency - I'm not familiar with the Airspy, but that probably means you are partially filtering your signal.

I'd suggest putting the device centre frequency at 162,450kHz and then add a 100kHz frequency offset to the NFM Demodulator (in the delta f control)

mistergitj, 2022-05-17:
Thanks, srcejon.  I will give this a try.  Enjoy, John.

mistergitj, 2022-05-17:
That works quite well, srcejon.  I have to do some more testing because I do not understand yet.  Thanks, John.

mistergitj, 2022-05-19:
On the Air Spy site for their R1/R2 products, I found this statement:
"No IQ imbalance, DC offset or 1/F noise at the center of the spectrum that plagues all the other SDRs"
I think this is telling me that I do not need the DC function to be set and conflicts with the HW.  I apologize that I missed this point.  Thanks again for the help offered.   Enjoy, John.

closed 2022-05-19 by mistergitj