f4exb/sdrangel #1250: Create default workspace in GUI

srcejon, 2022-05-18:
Would it be a little nicer for first-time users if a default workspace was created and default device opened the first time SDRangel is run (Perhaps with auto-stack layout enabled too - so it starts roughly like v6 used to)?

Just trying Ubuntu 22 for one of the other bugs, and when SDRangel first starts, we just get a tiny window like this:

[image]

f4exb, 2022-05-18:
Having a default setup at startup was a pain previously so I am not keen in reproducing errors from the past. Also once you have set up your first workspace it will be retrieved at the next startup so this is only a one time setup and later you will get used in the new way of working. Also the Quick start takes you step by step in creating your very first receiver. Maybe a mention of the auto-stack can be added.

I haven't tried 22.04 yet but I will upgrade my laptop in the coming days since I run in the soon to be phased out 21.10. Maybe this is due to a new release of Qt? The window is so tiny that I cannot check the Qt version :smile:

srcejon, 2022-05-18:
> Having a default setup at startup was a pain previously so I am not keen in reproducing errors from the past. Also once you have set up your first workspace it will be retrieved at the next startup [...]

It is in the docs - but you know what people are like when it comes to reading them! 

How about just creating a workspace without the device - so it's then just one click in the toolbar add the type of device they want? I'd imagine the basic user probably wont want to worry about workspaces. 

Looks clean enough to add, in mainwindow.cpp:

--- a/sdrgui/mainwindow.cpp
+++ b/sdrgui/mainwindow.cpp
@@ -202,6 +202,7 @@ MainWindow::MainWindow(qtwebapp::LoggerWithFile *logger, const MainParser& parse
     else
     {
         qDebug() << "MainWindow::MainWindow: scratch mode: do not load current configuration";
+        addWorkspace();
     }
 
     splash->showStatusMessage("finishing...", Qt::white);

And also gives a reasonable sized initial window.

>The window is so tiny that I cannot check the Qt version

Yeah, it's 5.15.3

srcejon, 2022-05-18:
On a related issue - is it possible to save the state of the auto-stack check button in the config somewhere?

srcejon, 2022-05-18:
Oops - yeah you're right. This seems to work, but maybe a bit hacky:

@@ -195,9 +195,18 @@ MainWindow::MainWindow(qtwebapp::LoggerWithFile *logger, const MainParser& parse
 
     if (!parser.getScratch())
     {
-        splash->showStatusMessage("load current configuration...", Qt::white);
-        qDebug() << "MainWindow::MainWindow: load current configuration...";
-        loadConfiguration(m_mainCore->m_settings.getWorkingConfiguration());
+        QSettings s;
+        if (s.contains("current-configuration"))
+        {
+            splash->showStatusMessage("load current configuration...", Qt::white);
+            qDebug() << "MainWindow::MainWindow: load current configuration...";
+            loadConfiguration(m_mainCore->m_settings.getWorkingConfiguration());
+        }
+        else
+        {
+            qDebug() << "MainWindow::MainWindow: no current configuration, creating empty workspace...";
+            addWorkspace();
+        }
     }

Up to you :)

On a slightly different subject, if you want to add a Mac build to the Assets on Releases page, I've put a DMG [here](http://sdrangel.org/downloads/sdrangel-7.0.0-macos.dmg):

f4exb, 2022-05-19:
Re Ubuntu 22 I have just upgraded to 22.04 and haven´t noticed any change in behaviour. However I have recompiled entirely by doing `touch ../CmakeLists.txt` before make. Also if you use the Python scripts in the `scriptsapi` folder you will have to re-create the virtual environment and do `pip install -r requirements.txt` inside this virtual environment.

f4exb, 2022-05-19:
Implemented in v7.0.1

closed 2022-05-19 by f4exb

alphafox02, 2022-05-19:
Will try this now as I also had a tiny window upon initial opening in 20.04.

f4exb, 2022-05-20:
the save/restore auto-stack is broken in v7.0.1 take the latest fix on master

alphafox02, 2022-05-20:
Any chance to make a deb update? Probably asking a lot for just one small change, but I’d change out 7.0.1 to 7.0.2 in my build ASAP.

f4exb, 2022-05-20:
All builds are listed here https://ci.appveyor.com/project/f4exb/sdrangel/history you can find artifacts by digging down in each build. The latest (as of now) is here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/43606020/job/nwy2ou4cngmcc464/artifacts

alphafox02, 2022-05-20:
Excellent thank you! I’ll bookmark this. Pulling down, installing, and that should finish the image I’m making.