f4exb/sdrangel #1255: SDRANGEL fail to start

sv1cdn, 2022-05-24:
Greetings from Athens!
Installed for the first time SDRANGEL version 7.1.0 in my Windows 10 Home version 21H1.
Unfortunately for me the sdrangel.exe fails to start no matter what I have tried. Microsoft Event Viewer reports the following:
  Description:
  Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x628b1202
  Faulting module name: **libusb-1.0.dll**, version: 1.0.23.11397, time stamp: 0x5e113705
  Exception code: **0xc0000005**
  Fault offset: 0x0000000000012ce3
  Faulting process id: 0x1a38
  Faulting application start time: 0x01d86f8a8a3c01e5
  Faulting application path: C:\Program Files\SDRangel\sdrangel.exe
  Faulting module path: **C:\Program Files\SDRangel\libusb-1.0.dll**
  Report Id: 79bcc72d-b062-4169-a1f1-77e06d189161
  Faulting package full name: 
  Faulting package-relative application ID: 

Something is wrong with the libusb-1.0.dll and my workstation setup here.
How can this be further troubleshooted and resolved please?
Thank you in advance!

73 de SV1CDN, Dennis!

srcejon, 2022-05-24:
What SDR do you have?

Maybe try unplugging USB devices and restarting sdrangel to see if the crash is related to a particular USB device.

sv1cdn, 2022-06-01:
For what is worth, IIO finally got up and running!

Had to replace libusb-1.0.dll with version 1.0.26.11724 from team's sourceforge space: https://sourceforge.net/projects/libusb/

sv1cdn, 2022-06-27:
> Can you please let us know if 7.3.2 fixes the issue for you?
>
> https://github.com/f4exb/sdrangel/releases

Greetings from Athens!Busy with QRL, sorry.
Do you still want me to try? In that case, guide me please. Do I need to uninstall current SDRANGEL version from my workstation?
73 de SV1CDN, Dennis!

srcejon, 2022-06-28:
Yes. Uninstall current version, and then install the latest here: https://github.com/f4exb/sdrangel/releases/download/v7.4.0/sdrangel-7.4.0-win64.exe

sv1cdn, 2022-07-06:
> Resending
>
> 73 de SV1CDN, Dennis!
>
> Greetings!I am sorry it took me so long in time to make it happen. Trying to balance work - family and hobby. Yes, version 7.4.0 works UFB! Didn't have to change any USB DLL like before.One question, easy one probably. How can I make fonts in W10 larger? Check screenshot, they are all way too cramped for me.
> TKS es stay safe!
> 73 de SV1CDN, Dennis!

Resending 

73 de SV1CDN, Dennis!

closed 2022-08-27 by srcejon