f4exb/sdrangel #1258: SDRangel the DMG way: no Funcube

loewal, 2022-05-26:
Is there a reason for no support for Funcube Pro+ ?

srcejon, 2022-05-26:
Yep, it needs HIDAPI, which isn't built.

f4exb, 2022-05-26:
Yes hidapi is required since #1160

srcejon, 2022-05-27:
There's a new dmg for 7.2.1 that should include the HIDAPI and Funcube plugin. I don't have one so can't test it.

loewal, 2022-05-27:
It is present Jon! Nice!

But...activating it gives no response...Tried Cubicsdr...works...badly.
GQRX? No.

I don't remenber if it ever worked as expected..

No visual signal. I wonder why RF filt is present....

Sorry that I wished the implementation!
Going to try on Win11

[image]
[image]

loewal, 2022-05-27:
Runs on SDR# (so..Windows) and SDRuno with EXTIO.dll. 

Noticed that v7.2.1 on Windows also misses the Funcube.
Installed with Macports and included Funcube Dongle Pro+, also no output..
Tested on Ubuntu, same as MacOs, nothing.

Took into account the udev etc.

srcejon, 2022-05-27:
> > Noticed that v7.2.1 on Windows also misses the Funcube.
> 
> HIDAPI probably needs to be added to sdrangel-windows-libraries.

Added a PR for those here: https://github.com/f4exb/sdrangel-windows-libraries/pull/12