f4exb/sdrangel #1259: M17 mod/demod

sp5wwp, 2022-05-27:
How about adding some support for the [M17 mode](https://m17project.org)?

abraxasneo, 2022-05-27:
I am a big fan of the M17 and such a receiver and transmitter implemented in your sdrangel would be something great. Because nobody has made a transmitter so far. There is only a receiver/decoder plugin for SDR++.  
M17 is an amateur alternative to commercial products like D-STAR, C4FN or DMR. Work is underway to implement this codec in the most popular transceivers such as the TYT MD-UV380.  
With your sdrangel, users could build a simple M17 transceiver and use it even for communications via the qo100 geostationary satellite.
I personally managed to build a simple transceiver from the command line on a raspberry pi and using the right syntax and the rpitx program, which transcodes the speech from the connected microphone in real time and transmits on a specific frequency. Based on this, I also sent a transmitter suspended under a stratospheric balloon, which transmitted the voice signal and in the digital part transmitted the height of the balloon to Earth. I am sure this is just the beginning of the possibilities of this M17 project.

f4exb, 2022-05-29:
Could be interesting indeed.

f4exb, 2022-06-03:
https://github.com/mobilinkd/m17-cxx-demod

f4exb, 2022-06-04:
NFM modulator needs more options to disable high pass filtering and maybe pre-emphasis.

sp5wwp, 2022-06-13:
This is how M17 RF spectrum should look like:

[[image]](https://en.wikipedia.org/wiki/M17_(amateur_radio))

f4exb, 2022-06-18:
What is your FM deviation? From the picture the base of the spectrum occupies roughly a third of the span which makes it ~16.7 kHz at -60 dB. It looks like it is obtained with a deviation much larger than the specified +/- 2.4 kHz. If I use +/-2.4 kHz I obtain something more narrow and spiky:
[image]

sp5wwp, 2022-06-18:
2.4 kHz. What's the baseband sample you use to get this?

> Formatting modulation file to be injected in NFM modulator: `sox /home/f4exb/f4exb_call.wav -t raw -r 8k -c 1 -b 16 -L -e signed - | apps/m17-mod -S F4EXB -D F5SFU | sox -r 48k -c 1 -b 16 -L -e signed -t raw - -t raw -r 48k -c 1 -b 32 -L -e float m17-sdrangel.raw`

This one? Can you share the resulting `.raw` file?

f4exb, 2022-06-18:
It is here now: https://transfert.free.fr/q2t84g
Note that I am using the m17-mod of https://github.com/mobilinkd/m17-cxx-demod so you should be able to play the same way with it. One thing I noticed is that the coefficients of the rrc filter are not those I obtain for aplha=0.5 Ts=1/4800 and Fs=48000 which should correspond to recommendations. However it does not seem to make a big difference.

sp5wwp, 2022-06-18:
This is what I get with GNU Radio for your baseband file:

[image]

The baseband generated by `cxx-demod` tools ***isn't*** an IQ stream, but rather an instantaneous frequency shift.

closed 2022-07-18 by f4exb

f4exb, 2022-07-20:
Due to too many issues the M17 plugins will be removed from the default build.