f4exb/sdrangel #1260: Satellite tracker crashes

srs4511351, 2022-05-29:
I am having Satellite tracker problems on SDRangel running on Raspberry Pi OS Bullseye 64 bit or DietPi Bullseye 64 bit. Same on both OS distributions.

Starting with version 6.19,
I added presets for the satellites that I have in Satellite tracker and configured SDRangel Control.
SDRangel will freeze when SDRangel Control starts a satellite path procedure while another Channel demod is running.
It freezes less often when there is nothing running between satellite passes.

The latest version that I tried with Satellite tracking working well was version 6.18.1.

Starting with version 6.19, in addition to the Satellite tracker crashes, the DSD demod constellation will freeze after running for a few minutes, sometimes longer.

I tried SDRangel version 7.2.1-8-g93e51e140
Satellite tracker/SDRangel control did not load the preset that I had set for the satellite when it became visible. Speech is working, as it announced the satellite becoming visible.
I closed the SSB demod that was running and opened APT. Satellite tracker was already running.
I set the frequencies and pressed play. SDRangel quit with this error:
2022-05-28 12:21:39.674 (D) APTDemodImageWorker::calcCoord: No TLE for satellite. Is Satellite Tracker running?
Segmentation fault

I restarted SDRangel, but it crashed again. The next time I ran SDRangel, it did not quit with Segmentation fault and I received the satellite.

I can provide more debug data if you tell me what and how to get it.

----Steve

srcejon, 2022-05-29:
Would be useful if you could let us know exactly which demods you have open. A screenshot might help.

If you want to debug, you can do:

    gdb /path/to/sdrangel
    run

Then when it crashes

    info stack 

And copy and paste the output here.

srs4511351, 2022-05-29:
The demods were SSB or DSD. Either one will do it. NFM may occasionally cause the crash, but the satellite presets included that. 

I can try the debug info when I have time.

——Steve

srs4511351, 2022-05-30:
I ran SDRangel 7.2.1-8-g93e51e140 under debug with the usual switches and to my surprise, it worked much better than it had been working. The only problem was that the spectrum/waterfall display froze.

gdb /opt/install/sdrangel/bin/sdrangel
run --soapy --fftwf-wisdom ~/.config/f4exb/fftw-wisdom

Was listening to SSB, when NOAA 18 became available.
SSB closed and APT opened.
ISS became available while NOAA 18 was visible. APT closed and NFM opened.
After ISS was no longer available, NFM closed and NOAA 18 opened APT again and received the satellite.

I received NFM and DSD for about 30 minutes.

I changed to SSB demod and closed NFM, but I left DSD Demod running. DSD Constellation did not freeze, but after about 30 minutes, the spectrum and waterfall display froze with SSB demod still receiving a signal.
I quit SDRangel and typed info stack.

#0  0x0000007ff5eb2e90 in read () at /lib/aarch64-linux-gnu/libc.so.6
#1  0x0000007ff5e5f5b8 in _IO_file_underflow ()
    at /lib/aarch64-linux-gnu/libc.so.6
#2  0x0000007ff5e52c48 in getdelim () at /lib/aarch64-linux-gnu/libc.so.6
#3  0x0000007f96d0ee1c in  () at /lib/aarch64-linux-gnu/libspeechd.so.2
#4  0x0000007f96d0f1e0 in  () at /lib/aarch64-linux-gnu/libspeechd.so.2
#5  0x0000007ff537d628 in start_thread ()
    at /lib/aarch64-linux-gnu/libpthread.so.0
#6  0x0000007ff5ec101c in  () at /lib/aarch64-linux-gnu/libc.so.6

srs4511351, 2022-06-02:
For some reason, SDRangel have been behaving much better that it had been.
In the past, 6.19 and newer crashed more often than not when Satellite tracker loaded a satellite preset.

V7 crashed frequently before I installed gdb. One strange thing that happened is that DSD demod constellation display never froze while running under gdb, but often did without gdb.

With that in mind, I booted up a Raspberry Pi OS with SDRangel 6.20.2. It does not have gdb installed.
With SDRangel 6.20.2, I had Satellite tracker, Maps, NFM, DSD running. Added SSB, there was no sound. Tried to close SSB and SDRangel froze. Force quit and restarted SDRangel and it happened again. Added SSB before hitting play and starting Satellite tracker and SSB works.
With those demods running, Satellite tracker loaded APT and NFM, spoke the message, then froze on satellite acquisition.
The terminal was rapidly displaying:
2022-06-01 23:02:40.745 (C) AudioFifo::write: (NFMDemod [0:0]) overflow 16384 samples
2022-06-01 23:02:40.746 (D) NFMDemodSink::feed: 0/16384 audio samples written
2022-06-01 23:02:40.746 (D) NFMDemodSink::feed: m_audioSampleRate: 48000 m_channelSampleRate: 0
2022-06-01 23:02:40.775 (C) AudioFifo::write: (NFMDemod [0:0]) overflow 16384 samples
2022-06-01 23:02:40.776 (D) NFMDemodSink::feed: 0/16384 audio samples written
2022-06-01 23:02:40.776 (D) NFMDemodSink::feed: m_audioSampleRate: 48000 m_channelSampleRate: 0
2022-06-01 23:02:40.805 (C) AudioFifo::write: (NFMDemod [0:0]) overflow 16384 samples
2022-06-01 23:02:40.805 (D) NFMDemodSink::feed: 0/16384 audio samples written
2022-06-01 23:02:40.805 (D) NFMDemodSink::feed: m_audioSampleRate: 48000 m_channelSampleRate: 0
2022-06-01 23:02:40.835 (C) AudioFifo::write: (NFMDemod [0:0]) overflow 16384 samples
2022-06-01 23:02:40.835 (D) NFMDemodSink::feed: 0/16384 audio samples written
2022-06-01 23:02:40.836 (D) NFMDemodSink::feed: m_audioSampleRate: 48000 m_channelSampleRate: 0
Killed

After this, cannot start SDRangel:
[ERROR] ReleaseDevice Error: sdrplay_api_ServiceNotResponding

I hope this is useful.

srs4511351, 2022-06-03:
SDRangel 7.2.1-8-g93e51e140  running under gdb:
gdb /opt/install/sdrangel/bin/sdrangel
run --soapy --fftwf-wisdom ~/.config/f4exb/fftw-wisdom

Had been receiving NFM and DSD.
NOAA 15 became visible and it received the satellite image.
Durng this time, NOAA 19 became visible, but NOAA 15 continued to be received.
When NOAA 15 was no longer visible, SDRangel had a Segmentation fault, probably when switching to NOAA 19, which was still visible.

I kept this terminal output, including info stack near the end:

2022-06-02 19:31:07.740 (D) SatelliteTrackerWorker::doppler  "NOAA 15"
2022-06-02 19:31:07.741 (D) APTDemod::webapiSettingsPutPatch: forward to GUI: 0x5558d8b3d0
2022-06-02 19:31:07.781 (D) APTDemod::handleMessage: MsgConfigureAPTDemod
2022-06-02 19:31:07.782 (D) APTDemod::applySettings:  m_cropNoise:  false  m_denoise:  true  m_linearEqualise:  false  m_histogramEqualise:  false  m_precipitationOverlay:  false  m_flip:  true  m_channels:  0  m_decodeEnabled:  true  m_autoSave:  false  m_autoSavePath:  ""  m_autoSaveMinScanLines:  200  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2022-06-02 19:31:07.782 (D) APTDemodGUI::handleMessage: APTDemod::MsgConfigureAPTDemod
2022-06-02 19:31:07.783 (D) APTDemodImageWorker::handleMessage: MsgConfigureAPTDemodImageWorker
2022-06-02 19:31:07.784 (D) APTDemodBaseband::handleMessage
2022-06-02 19:31:07.784 (D) APTDemodBaseband::handleMessage: MsgConfigureAPTDemodBaseband
2022-06-02 19:31:07.784 (D) DownChannelizer::applyChannelization done:  nb stages: 4  in (baseband): 2000000  req: 48000  out (channel): 125000  fc: -7630
2022-06-02 19:31:07.784 (D) APTDemodSink::applyChannelSettings:  channelSampleRate:  125000  channelFrequencyOffset:  -7630
2022-06-02 19:31:07.784 (D) NCO freq: 7630.000000 phase inc 250
2022-06-02 19:31:07.784 (D) APTDemodSink::applySettings:  m_rfBandwidth:  40000  m_fmDeviation:  17000  m_decodeEnabled:  true  force:  false
2022-06-02 19:31:08.740 (D) SatelliteTrackerWorker::los  "NOAA 15"  target:  "NOAA 15"
2022-06-02 19:31:08.741 (D) APTDemod::webapiActionsPost - LOS  "NOAA 15"
2022-06-02 19:31:08.741 (D) SatelliteTrackerWorker::los - autoTarget setting  "NOAA 19"
2022-06-02 19:31:08.741 (D) SatelliteTrackerWorker::aos: Loading preset  "NOAA 19"  to device set at  0
2022-06-02 19:31:08.754 (D) SatelliteTracker::handleMessage: MsgConfigureSatelliteTracker
2022-06-02 19:31:08.754 (D) SatelliteTracker::applySettings:  m_latitude:  42.2336  m_longitude:  -83.5802  m_heightAboveSeaLevel:  227  m_target:  "NOAA 19"  m_satellites:  ("ISS", "NOAA 15", "NOAA 18", "NOAA 19")  m_tles:  ("https://db.satnogs.org/api/tle/", "https://www.amsat.org/tle/current/nasabare.txt", "https://www.celestrak.com/NORAD/elements/goes.txt")  m_dateTime:  ""  m_minAOSElevation:  5  m_minPassElevation:  15  m_azElUnits:  1  m_groundTrackPoints:  100  m_dateFormat:  "yyyy/MM/dd"  m_utc:  false  m_updatePeriod:  1  m_dopplerPeriod:  10  m_defaultFrequency:  1e+08  m_drawOnMap:  true  m_autoTarget:  true  m_aosSpeech:  "${name} is visible for ${duration} minutes. Max elevation, ${elevation} degrees."  m_losSpeech:  "${name} is no longer visible."  m_aosCommand:  ""  m_losCommand:  ""  m_predictionPeriod:  5  m_passStartTime:  QTime("00:00:00.000")  m_passFinishTime:  QTime("23:59:59.000")  m_deviceSettings:  QHash(("NOAA 15", 0x555692fb20)("NOAA 18", 0x5556c52bd0)("NOAA 19", 0x5556938c20)("ISS", 0x5556928870))  m_title:  "Satellite Tracker"  m_rgbColor:  4292942179  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIFeatureSetIndex:  0  m_reverseAPIFeatureIndex:  0  force:  false
2022-06-02 19:31:08.755 (D) SatelliteTrackerWorker::applySettings:  m_target:  "NOAA 19"  m_satellites:  ("ISS", "NOAA 15", "NOAA 18", "NOAA 19")  m_dateTime:  ""  m_utc:  false  m_updatePeriod:  1  force:  false
2022-06-02 19:31:08.756 (D) APTDemod::handleMessage: MsgConfigureAPTDemod
2022-06-02 19:31:08.756 (D) APTDemod::applySettings:  m_cropNoise:  false  m_denoise:  true  m_linearEqualise:  false  m_histogramEqualise:  false  m_precipitationOverlay:  false  m_flip:  true  m_channels:  0  m_decodeEnabled:  false  m_autoSave:  false  m_autoSavePath:  ""  m_autoSaveMinScanLines:  200  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  false
2022-06-02 19:31:08.756 (D) APTDemodGUI::handleMessage: APTDemod::MsgConfigureAPTDemod
2022-06-02 19:31:08.756 (D) APTDemodImageWorker::handleMessage: MsgConfigureAPTDemodImageWorker
2022-06-02 19:31:08.757 (D) APTDemodBaseband::handleMessage
2022-06-02 19:31:08.757 (D) APTDemodBaseband::handleMessage: MsgConfigureAPTDemodBaseband
2022-06-02 19:31:08.758 (D) APTDemodSink::applySettings:  m_rfBandwidth:  40000  m_fmDeviation:  17000  m_decodeEnabled:  false  force:  false
2022-06-02 19:31:08.759 (D) SDRPlayV3Input::handleMessage: MsgStartStop:  stop
2022-06-02 19:31:08.759 (D) DSPDeviceSourceEngine::stopAcquistion
2022-06-02 19:31:08.760 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-06-02 19:31:08.760 (D) DSPDeviceSourceEngine::gotoIdle
2022-06-02 19:31:08.760 (D) SDRPlayV3Input::stop
2022-06-02 19:31:09.096 (D) GLSpectrumGUI::handleInputMessages: message: SpectrumVis::MsgStartStop
2022-06-02 19:31:09.096 (D) APTDemod::stop
[... cut ...]
#16 0x0000007ff64148ac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#17 0x0000007ff62299ec in QThread::exec() ()
    at /lib/aarch64-linux-gnu/libQt5Core.so.5
#18 0x0000007ff622ae6c in  () at /lib/aarch64-linux-gnu/libQt5Core.so.5
#19 0x0000007ff537d628 in start_thread ()
    at /lib/aarch64-linux-gnu/libpthread.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#20 0x0000007ff5ec101c in  () at /lib/aarch64-linux-gnu/libc.so.6
(gdb)

srcejon, 2022-06-05:
The crash there appears to be in the APT demodulator - you might want to try running without that demod for now.

srs4511351, 2022-06-07:
Thanks for looking at this. I hope that you will fix this when you have time.
I still love SDRangel.

----Steve

closed 2022-08-12 by github-actions[bot]

srs4511351, 2022-08-12:
I am hoping that you will be able to fix it when you get time.
Thanks, Steve

srs4511351, 2022-10-11:
I am hoping that you will be able to fix APT.
This issue serves as a reminder.

----Steve