f4exb/sdrangel #1263: Reasons for 'boost::chrono'

gvanem, 2022-05-30:
I'm curious of why `boost::chrono` is needed in e.g. `plugins/channelrx/demodadsb`.

I believe it was once true that on Windows, the `stl::steady_clock` was rather unprecise (using `ClockTick()`?)
compared to `boost::chrono`. But [today](https://docs.microsoft.com/en-us/cpp/standard-library/steady-clock-struct?view=msvc-170), it uses `QueryPerformanceCounter()`.

So with these patches and a `-DUSE_STL_CHRONO` in my `CFLAGS`, it builds fine:
```diff
--- a/adsbdemod.cpp 2022-05-20 20:59:47
+++ b/adsbdemod.cpp 2022-05-30 10:21:59
@@ -17,9 +17,6 @@
 // along with this program. If not, see <http://www.gnu.org/licenses/>.          //
 ///////////////////////////////////////////////////////////////////////////////////

-#define BOOST_CHRONO_HEADER_ONLY
-#include <boost/chrono/chrono.hpp>
-
 #include <stdio.h>
 #include <complex.h>

--- a/adsbdemodsink.cpp 2022-05-05 11:34:05
+++ b/adsbdemodsink.cpp 2022-05-30 10:23:28
@@ -61,7 +61,7 @@
 void ADSBDemodSink::feed(const SampleVector::const_iterator& begin, const SampleVector::const_iterator& end)
 {
     // Start timing how long we are in this function
-    m_startPoint = boost::chrono::steady_clock::now();
+    m_startPoint = BOOST(chrono::steady_clock::now());

     // Optimise for common case, where no resampling or frequency offset
     if ((m_interpolatorDistance == 1.0f) && (m_channelFrequencyOffset == 0))
@@ -122,7 +122,7 @@
     }

     // Calculate number of seconds in this function
-    boost::chrono::duration<double> sec = boost::chrono::steady_clock::now() - m_startPoint;
+    BOOST(chrono::duration<double>) sec = BOOST(chrono::steady_clock::now()) - m_startPoint;
     m_feedTime += sec.count();
 }

@@ -148,13 +148,13 @@
             m_writeBuffer = 0;

         // Don't include time spent waiting for a buffer
-        boost::chrono::duration<double> sec = boost::chrono::steady_clock::now() - m_startPoint;
+        BOOST(chrono::duration<double>) sec = BOOST(chrono::steady_clock::now()) - m_startPoint;
         m_feedTime += sec.count();

         if (m_worker.isRunning())
             m_bufferWrite[m_writeBuffer].acquire();

-        m_startPoint = boost::chrono::steady_clock::now();
+        m_startPoint = BOOST(chrono::steady_clock::now());

         m_writeIdx = m_samplesPerFrame - 1; // Leave space for copying samples from previous buffer

--- a/adsbdemodsink.h 2022-05-05 11:34:05
+++ b/adsbdemodsink.h 2022-05-30 10:30:52
@@ -19,14 +19,20 @@
 #ifndef INCLUDE_ADSBDEMODSINK_H
 #define INCLUDE_ADSBDEMODSINK_H

-#define BOOST_CHRONO_HEADER_ONLY
-#include <boost/chrono/chrono.hpp>
-
 #include "dsp/channelsamplesink.h"
 #include "dsp/nco.h"
 #include "dsp/interpolator.h"
 #include "util/movingaverage.h"

+#if defined(USE_STL_CHRONO)
+  #include <chrono>
+  #define BOOST(...)  std::__VA_ARGS__
+#else
+  #define BOOST_CHRONO_HEADER_ONLY
+  #include <boost/chrono/chrono.hpp>
+  #define BOOST(...)  boost::__VA_ARGS__
+#endif
+
 #include "adsbdemodsettings.h"
 #include "adsbdemodstats.h"
 #include "adsbdemodsinkworker.h"
@@ -85,7 +91,7 @@
     Real m_interpolatorDistance;
     Real m_interpolatorDistanceRemain;

-    boost::chrono::steady_clock::time_point m_startPoint;
+    BOOST(chrono::steady_clock::time_point) m_startPoint;
     double m_feedTime;                  //!< Time spent in feed()

     // Triple buffering for sharing sample data between two threads

--- a/adsbdemodsinkworker.cpp 2022-05-05 11:34:05
+++ b/adsbdemodsinkworker.cpp 2022-05-30 10:19:42
@@ -15,9 +15,6 @@
 // along with this program. If not, see <http://www.gnu.org/licenses/>.          //
 ///////////////////////////////////////////////////////////////////////////////////

-#define BOOST_CHRONO_HEADER_ONLY
-#include <boost/chrono/chrono.hpp>
-
 #include <QDebug>

 #include "util/stepfunctions.h"
@@ -42,7 +39,7 @@
     m_sink->m_bufferRead[readBuffer].acquire();

     // Start recording how much time is spent processing in this method
-    boost::chrono::steady_clock::time_point startPoint = boost::chrono::steady_clock::now();
+    BOOST (chrono::steady_clock::time_point startPoint = BOOST(chrono::steady_clock::now()));

     // Check for updated settings
     handleInputMessages();
@@ -271,7 +268,7 @@
                 nextBuffer = 0;

             // Update amount of time spent processing (don't include time spend in acquire)
-            boost::chrono::duration<double> sec = boost::chrono::steady_clock::now() - startPoint;
+            BOOST(chrono::duration<double>) sec = BOOST(chrono::steady_clock::now()) - startPoint;
             m_demodStats.m_demodTime += sec.count();
             m_demodStats.m_feedTime = m_sink->m_feedTime;

@@ -291,7 +288,7 @@
                 handleInputMessages();

                 // Resume timing how long we are processing
-                startPoint = boost::chrono::steady_clock::now();
+                startPoint = BOOST(chrono::steady_clock::now());

                 int samplesRemaining = m_sink->m_bufferSize - readIdx;
                 if (samplesRemaining > 0)
```

Or would it make sense to apply such a diff to `sdrbase/util/timeutil.cpp`  and simply
use functions from there instead?

srcejon, 2022-05-30:
As boost is used elsewhere, why not use it?

gvanem, 2022-05-30:
Elsewhere yes. But looking at the use of Boost a bit deeper, I see that in these files:
 *  `plugins/channelrx/filesink/filesink.cpp`
 * `plugins/channelrx/localsink/localsink.cpp`  
 * `plugins/channelrx/localsink/ localsinksink.cpp`
  
these Boost headers are not needed:
```c
 #include <boost/crc.hpp>
 #include <boost/cstdint.hpp>
```
Perhaps an old patch made these redundant?

So that leaves the use of `boost::format()`, `boost::math::barycentric_rational()` etc.
Could these be replaced with some Qt-functions?

I'm just asking. (I do have Boost 1.79.0 installed).

srcejon, 2022-05-30:
Of course, you can rewrite anything - but I don't see the benefit of doing so. Boost is also needed for several of the 3rd party libraries.

gvanem, 2022-05-30:
>  Boost is also needed for several of the 3rd party libraries.

Besides UHD, which one?

srcejon, 2022-05-30:
I haven't looked closely, but it looks like cm256cc.

gvanem, 2022-05-30:
>  but it looks like cm256cc.

Not after [this](https://github.com/f4exb/cm256cc/pull/15) was merged.

closed 2022-08-04 by github-actions[bot]