f4exb/sdrangel #1264: De-allocating unused source receivers from inactive workspaces in the OS

Hermite123456, 2022-05-30:
Question: 
I have 5 workspaces and 5 RTL-SDR dongles. If I start 1 workspace receiver and not the other 4, the source receivers within the other workspace remain allocated to the operating system (Windows) although I did not even start them. When not in use, I would lilke to be able to use these other 4 instances with an external application, but they remain hogged by SDRAngel. Is there a workaround, or, by design it will remain as such? Thank you.

f4exb, 2022-05-30:
It is not a matter of workspaces but a matter of device sets (or receivers). If you create 5 such receivers for your 5 RTL-SDRs then they will be allocated to SDRangel and not available to other applications it does not matter if the device is started or stopped.

Hermite123456, 2022-05-31:
OK, understood. Perhaps a future enhancement where we can 'Lock/Release' individual receivers. I know easier said than done :) Thank you.

closed 2022-05-31 by Hermite123456