f4exb/sdrangel #1266: Funcube

loewal, 2022-05-30:
Neither latest Macos-version, selfcompiled, nor latest Windows version, supplied by Jon, works.
Even compiled on Linux Jellyfish.
I started fcdhid to check presence of the dongle. It was recognized.

It works nowhere....I think it never worked.
Why does SDRuno with extio.dll work? Also SDR# (not fluently..)?

f4exb, 2022-05-30:
> It works nowhere....I think it never worked.

Really?
[image]

f4exb, 2022-05-31:
I tried the Windows version. I think that the problem is that it does not recognize the audio device of the FunCube. The IQ data of the FunCube is sampled via an audio device. As a consequence the device runs but yields no data. The controls (hence hidapi) seem to work (no errors).

loewal, 2022-05-31:
Strange Edouard,
I noticed several things:
- mix is not enabled
- RF filt differs from frequency: mine is 192k
- I see nog signal at all

Are there many others with these problems?
Otherwise, remove Funcube form SDRA.

[image]

loewal, 2022-05-31:
Must correct myself:
- above settings are in MacOs
- in Win11 the samplerate is 96K; shouldn't it be the fixed 192K?
- rest: no signal

f4exb, 2022-05-31:
> Otherwise, remove Funcube form SDRA.

Works perfectly in Linux there is no reason to remove it.

loewal, 2022-05-31:
Installed also in Linux (Jellyfish).
My udev-rules:
# Udev rules for the Funcube Dongle Pro+ (0xfb31) 

# HIDAPI/libusb: 

SUBSYSTEMS=="usb", ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="fb31", GROUP:="users", MODE:="0660", SYMLINK+="FCDPP" 

# HIDAPI/hidraw: 

#KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="fb31", MODE="0666" 

Still no signal.
Device works in Windows: SDRuno and SDR#.

Is there perhaps something I could have overlooked?
When started from the commandprompt I see, that the Halincrest Funcube v2.0-analog-stereo is the input....

Is that alright?

srcejon, 2022-05-31:
> I tried the Windows version. I think that the problem is that it does not recognize the audio device of the FunCube. The IQ data of the FunCube is sampled via an audio device. 

In FCDProInput::openFCDAudio it looks for an audio device with a specific name, which comes from fcdlib/fcdtraits.cpp, which can be either:

    const char *fcd_traits<Pro>::qtDeviceName = "FUNcube_Dongle_V1.0";
    const char *fcd_traits<ProPlus>::qtDeviceName = "FUNcube_Dongle_V2.0";

Do you have an audio device with that name?

Do you get   qCritical("FCDProInput::openFCDAudio: device with name %s not found", cardname); printed in your log? That should be printed if it fails to find the device.

f4exb, 2022-05-31:
I can see it in the sound devices in the Devices manager:
[image]
but I cannot see it in the audio devices in SDRangel:
[image]

f4exb, 2022-06-01:
OK for me now on Windows with b5ad1b27f5ef0356d002c695e294cdccda8fa412
[image]

loewal, 2022-06-01:
Awaiting this version on Github!

f4exb, 2022-06-01:
It is there already (unofficial): https://github.com/f4exb/sdrangel/actions/runs/2418252820

loewal, 2022-06-01:
Hi Edouard,

Beautiful!!
At last it’s working here too!

Thank you very much.

Grtz.

srcejon, 2022-06-01:
> Hi Edouard, Beautiful!! At last it’s working here too! Thank you very much. Grtz.

It might help if you can post a screenshot of Audio Devices > Audio Inputs from MacOS, so we can see if it's the same problem.

loewal, 2022-06-01:
I managed to get things working on an old Dell laptop on Linux 20.04.
The problems I found were on an 10 year old iMac 27" running Jammy Jellyfish.
Perhaps the namingconventions are therefore different...?

loewal, 2022-06-01:
[image]
[image]

loewal, 2022-06-01:
This on my iMac 27" with Jammy Jellyfish:
[image]

f4exb, 2022-06-01:
It looks similar on Mac and Windows so Linux being the odd guy there by reversing the condition it should work for all 3 systems.

f4exb, 2022-06-02:
Implemented in v7.3.1

closed 2022-06-02 by f4exb