f4exb/sdrangel #1269: Noise Figure Plugin - won't run

Bob-Sellers, 2022-06-01:
Problem present on Windows 7, SDRAngel 7.2.1

After performing workaround to allow running of PSU control script (instead of VISA / SCPI PSU control), as mentioned in #1268, the Noise Figure Plugin doesn't appear to step through the frequencies and perform the Noise Figure measurement when pressing the 'play' button. 

The power meter in the Noise Figure plugin appears to be working correctly and responds appropriately to changes in receiver gain. However, the radio does not retune. I'm using the default frequency sweep settings (430 MHz to 440MHz in 3 steps).

I'm running Windows and I'm assuming that the PSU control script can be a batch file?

I have limited debug experience/capability but if I enable logging and try to start the Noise Figure measurement, I get the following log message, which may be relevant:

2022-05-31 11:59:15.568 (W) QMetaObject::invokeMethod: No such method ChannelAPI::nextState()

I get the same issue with both my SDR's, ADALM-PLUTO and SDRPlay.

Eventually, I would like to do this test on a networked USRP but understand there are some issues at present talking to an IP based USRP?

srcejon, 2022-06-01:
Looks like something has broken with the API changes made for v7. Should be straight forward to fix.

srcejon, 2022-06-01:
> Eventually, I would like to do this test on a networked USRP but understand there are some issues at present talking to an IP based USRP?

I've just added a patch for that as well: https://github.com/f4exb/sdrangel/pull/1271