f4exb/sdrangel #1272: AMBE Controller on Windows only gets failed frames using a DV Stick 30

Hermite123456, 2022-06-01:
Hello,
I'm on a NXDN-4800 frequency and I can hear normal audio decoded without the AMBE controller.
When the AMBE Controller is enabled (connected via BlueDV AMBE 3000 Server), I only get failed frames when decoding.
Am I doing something wrong ? (I'm all new to this, so bare with me:) )

Thank you.
[
[image]
](url)

f4exb, 2022-06-02:
Check your network setup. For me it works:
[image]

Note: using AMBE server from https://github.com/marrold/AMBEServer on a RPi4.

Hermite123456, 2022-06-02:
Ok, but do you (or anybody else) have a proven working connectivity example of an AMBE Server on Windows 10 ?
 
I'm using BlueDV and BlueDV AMBE Server and they work just fine on Windows 10. No IP:PORT conflicts.

Checking my Network setup is a broad statement... what could possibly be the culprit if BlueDV connectivity to AMBE Server works fine?

closed 2022-08-08 by github-actions[bot]