f4exb/sdrangel #1274: Freeze using 7.3.1 on Radio Astronomy.

mistergitj, 2022-06-03:
I do not like to open an issue that I have posted, but I feel I should have open one instead of posting in SDRAngel.  This continues to happen and needs a look.  It was suggested to post my log file and it is here.  Please let me know of any other information that will help. Thanks much and enjoy, John.
log file:
[sdrangel.log](https://github.com/f4exb/sdrangel/files/8828579/sdrangel.log)
Original post:
I am running 7.3.1 on W10 x64. Here is a screenshot.  After running for some time SDRAngel hangs.  I am forced to use the Task manager to terminate the 'Not Responding' task.
Please tell me what information to provide so that the developers can figure out what is causing this.  Here is a screenshot of the running job.
Thanks much, John.

mistergitj, 2022-06-03:
Screenshot is here:
https://groups.io/g/sdrangel/topic/freeze_using_7_3_1_on_radio/91504798?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,91504798,previd%3D1654241438214223856,nextid%3D1650138831891993541&previd=1654241438214223856&nextid=1650138831891993541

mistergitj, 2022-06-03:
W10 Eventlog for one hang:
`Log Name:      Application
Source:        Application Hang
Date:          6/1/2022 1:07:24 PM
Event ID:      1002
Task Category: (101)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      xxxxxxxxxxxxx
Description:
The program sdrangel.exe version 0.0.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Security and Maintenance control panel.
 Process ID: 25cc
 Start Time: 01d875decee4aa98
 Termination Time: 7
 Application Path: C:\Program Files\SDRangel\sdrangel.exe
 Report Id: 7aa3a2d8-7c3c-44fa-96ad-fc71b7440230
 Faulting package full name: 
 Faulting package-relative application ID: 
 Hang type: Unknown

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Hang" />
    <EventID Qualifiers="0">1002</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>101</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-06-01T18:07:24.1319335Z" />
    <EventRecordID>26057</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>xxxxxxxxx</Computer>
    <Security />
  </System>
  <EventData>
    <Data>sdrangel.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>25cc</Data>
    <Data>01d875decee4aa98</Data>
    <Data>7</Data>
    <Data>C:\Program Files\SDRangel\sdrangel.exe</Data>
    <Data>7aa3a2d8-7c3c-44fa-96ad-fc71b7440230</Data>
    <Data>
    </Data>
    <Data>
    </Data>
    <Data>Unknown</Data>
    <Binary>55006E006B006E006F0077006E0000000000</Binary>
  </EventData>
</Event>`

mistergitj, 2022-06-04:
Another W10 Event for a different module:
   f4exb/sdrangel
   Log Name:      Application
Source:        Application Error
Date:          6/3/2022 1:52:18 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      John-AMD-TR
Description:
Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x6297c3e2
Faulting module name: inputairspy.dll, version: 0.0.0.0, time stamp: 0x6297c6de
Exception code: 0xc0000005
Fault offset: 0x0000000000003172
Faulting process id: 0xf50
Faulting application start time: 0x01d8777adde46dd3
Faulting application path: C:\Program Files\SDRangel\sdrangel.exe
Faulting module path: C:\Program Files\SDRangel\plugins\inputairspy.dll
Report Id: db3b27cc-b09c-4495-9cf0-3495027532cb
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2022-06-03T18:52:18.1160519Z" />
    <EventRecordID>26187</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>??????</Computer>
    <Security />
  </System>
  <EventData>
    <Data>sdrangel.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>6297c3e2</Data>
    <Data>inputairspy.dll</Data>
    <Data>0.0.0.0</Data>
    <Data>6297c6de</Data>
    <Data>c0000005</Data>
    <Data>0000000000003172</Data>
    <Data>f50</Data>
    <Data>01d8777adde46dd3</Data>
    <Data>C:\Program Files\SDRangel\sdrangel.exe</Data>
    <Data>C:\Program Files\SDRangel\plugins\inputairspy.dll</Data>
    <Data>db3b27cc-b09c-4495-9cf0-3495027532cb</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

srcejon, 2022-06-05:
I think what is probably happening here, is that because you have such a small integration time (t sub sigma), after a while, it takes longer to process and display the results than it does to take the measurement and the GUI can't keep up.

You probably want to run with a higher integration count (sigma) anyway - as this increases the sensitivity of the measurement.

mistergitj, 2022-06-05:
Thanks, srcejon.  Another user recently suggested making sure the integration time is over 1.0s and I have been running that way since.  I do not think I have seen a hang since.  It would be nice if the program could get past the error a little more gracefully.  Are these types of problems due to my running Windows? Thanks again and enjoy, John.

srcejon, 2022-06-05:
I don't think the problem is Windows. I suspect it's just that you're asking it to do more processing than it can handle.

mistergitj, 2022-06-05:
Thanks, srcejon.  My processor is an AMD 1950X with 16 cores and 32 threads Threadripper. My system is on a RAID0 of SSDs if it makes any difference.  Thanks and enjoy, John.

srcejon, 2022-06-05:
>. My processor is an AMD 1950X with 16 cores and 32 threads Threadripper. 

The processing that is most likely causing a problem is all on a single thread at the moment - so extra cores not beneficial.

closed 2022-08-11 by github-actions[bot]