f4exb/sdrangel #1277: Hang on Radio Astronomy Channel when attempt to use 'Save charts to animation file'.dio Astromony

mistergitj, 2022-06-04:
I am running 7.3.1 under W10 x64.  When I have the Astronomy Channel running and try to use the camera Icon, SDRAngel hangs as soon as I click Save for the specified file.  If I stop the radio there is no hang but am only offered a .png file type.
Thanks much, John.

srcejon, 2022-06-05:
The animation is saved as an animated .png file, so that is the expected behaviour. If you need another file format, you'll need to use another program to convert from png to whatever you want.

I can't reproduce the hang at the moment.

mistergitj, 2022-06-05:
Thanks, srcejon. Are you running under Windows?  One thing I have noticed using SDRAngel (only a few weeks) is that it is a little fragile. Should the radio be stopped before changing any settings?  When I click Save after opening the camera icon on the Radio Astronomy channel Spectrum, SDRAngel almost always hangs at least for several seconds.  When it continues to run the target file grows but usually stops with no intervention.  Please tell me what steps to use to start and stop the camera icon function.  I have found that Firefox will open the .png animation file.  Could you please allow an .apng file type so I can assign the application to open it?  Is there a way to ascertain if the animated file is still being updated.  Sometimes I get a few 100 KB and other several MB?  It almost always seems to quit on it own. 
Thanks so much for your help.  Enjoy, John.

srcejon, 2022-06-05:
You shouldn't need to stop the radio while changing settings.

The save animation feature should only save the spectra that have been collected so far - it wont append new data. You probably want to stop any measurement before pressing save, so it's not trying to do two things at once. The size of the file will depend on how many spectra there are and how different they are.

closed 2022-08-11 by github-actions[bot]