f4exb/sdrangel #1278: USRP E320 - Overflows when adding channel

Bob-Sellers, 2022-06-05:
I'm greatly appreciating the capability to use an ethernet based USRP with SDRAngel but I've encountered a few issues, the first was that I needed to replace the uhd.dll in the SDRAngel folder with a newer version. (Upgraded from UHD 4.0 to UHD 4.2).

The main problem I'm encountering now is that when I add an SDRAngel 'channel' (Broadcast FM Demod) to process the received data, I invariably get overflows (signified in the log and with the red O indicator), this is for low data rates eg 1MS/s. From memory I'm sure I've had success with an RTL-SDR under similar conditions. The demod audio is very slow and distorted.

Without a 'channel' attached I can view the spectrum and waterfall of a channel sampled at a reported 5.333 333 MS/s without issue.

Bob-Sellers, 2022-06-05:
Tried to capture a log file, but in this instance the radio didn't appear to drop samples, just the audio device:

2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::initAcquisition
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionInit
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoIdle
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoInit:   m_deviceDescription:  USRPInput  sampleRate:  1000000  centerFrequency:  93000000
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoInit: initializing  SpectrumVis
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoInit: initializing  BFMDemod
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::startAcquisition
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStart
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoRunning
2022-06-05 14:58:03.420 (D) DSPDeviceSourceEngine::gotoRunning:  USRPInput  started
2022-06-05 14:58:03.420 (D) USRPInput::suspendRxBuddies (0)
2022-06-05 14:58:03.420 (D) USRPInput::suspendTxBuddies (0)
2022-06-05 14:58:03.423 (D) USRPInput::applySettings: clock set to  "internal"
2022-06-05 14:58:03.423 (D) USRPInput::applySettings: set sample rate set to 1000000
2022-06-05 14:58:03.535 (D) USRPInput::applySettings: frequency set to 93000000 with LO offset 0
2022-06-05 14:58:03.577 (D) USRPInput::applySettings: AGC enabled for channel  0
2022-06-05 14:58:03.577 (D) USRPInput::applySettings: set antenna path to TX/RX on channel 0
2022-06-05 14:58:03.584 (D) USRPInput::applySettings: actual sample rate 1000000.000000
2022-06-05 14:58:03.584 (D) USRPInput::applySettings: master_clock_rate 61440000.000000
2022-06-05 14:58:03.584 (D) USRPInput::applySettings: forward change to all buddies
2022-06-05 14:58:03.586 (D) USRPInput::applySettings: clock source is  "internal"
2022-06-05 14:58:03.586 (D) USRPInput::applySettings: center freq:  93000000  Hz  m_transverterMode:  false  m_transverterDeltaFrequency:  0  deviceCenterFrequency:  93000000  device stream sample rate:  1,000,000 S/s  sample rate with soft decimation:  1,000,000 S/s  m_log2SoftDecim:  0  m_gain:  50  m_lpfBW:  10,000,000  m_antennaPath:  TX/RX  m_clockSource:  internal  force:  true
2022-06-05 14:58:03.667 (D) USRPInput::resumeTxBuddies (0)
2022-06-05 14:58:03.667 (D) USRPInput::resumeRxBuddies (0)
2022-06-05 14:58:03.667 (D) USRPInput::start: thread created
2022-06-05 14:58:03.667 (D) USRPInputThread::issueStreamCmd  start
2022-06-05 14:58:03.667 (D) USRPInputThread::startWork: stream started
2022-06-05 14:58:03.668 (D) DSPDeviceSourceEngine::gotoRunning: starting  SpectrumVis
2022-06-05 14:58:03.668 (D) DSPDeviceSourceEngine::gotoRunning: starting  BFMDemod
2022-06-05 14:58:03.668 (D) BFMDemod::start
2022-06-05 14:58:03.668 (D) DownChannelizer::applyChannelization done:  nb stages: 0  in (baseband): 1000000  req: 0  out (channel): 1000000  fc: 0
2022-06-05 14:58:03.668 (D) BFMDemodSink::applyChannelSettings:  channelSampleRate:  1000000  channelFrequencyOffset:  0
2022-06-05 14:58:03.669 (D) NCO freq: 0.000000 phase inc 0
2022-06-05 14:58:03.669 (D) PhaseLock::configure: freq: 0.019000 bandwidth: 0.000050 minsignal: 0.010000
2022-06-05 14:58:03.669 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2022-06-05 14:58:03.669 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2022-06-05 14:58:03.669 (D) Interpolator::createPolyphaseLowPass: ntaps: 18
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::gotoRunning:input message queue pending:  1
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  1000000  m_centerFrequency:  93000000
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  BFMDemod
2022-06-05 14:58:03.670 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x122009b6b60
2022-06-05 14:58:03.673 (D) BFMDemodBaseband::handleMessage: MsgConfigureBFMDemodBaseband
2022-06-05 14:58:03.674 (D) DownChannelizer::applyChannelization done:  nb stages: 1  in (baseband): 1000000  req: 270000  out (channel): 500000  fc: 0
2022-06-05 14:58:03.674 (D) BFMDemodSink::applyChannelSettings:  channelSampleRate:  500000  channelFrequencyOffset:  0
2022-06-05 14:58:03.674 (D) NCO freq: 0.000000 phase inc 0
2022-06-05 14:58:03.674 (D) PhaseLock::configure: freq: 0.038000 bandwidth: 0.000100 minsignal: 0.010000
2022-06-05 14:58:03.674 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2022-06-05 14:58:03.674 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2022-06-05 14:58:03.675 (D) Interpolator::createPolyphaseLowPass: ntaps: 18
2022-06-05 14:58:03.675 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  93000000  sampleRate:  1000000
2022-06-05 14:58:03.675 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  93000000  sampleRate:  1000000
2022-06-05 14:58:03.675 (D) BFMDemodSink::applyAudioSampleRate: 48000
2022-06-05 14:58:03.675 (D) BFMDemod::handleMessage: DSPSignalNotification
2022-06-05 14:58:03.675 (D) Interpolator::createPolyphaseLowPass: ntaps: 72
2022-06-05 14:58:03.676 (D) BFMDemodGUI::handleMessage: BFMDemodReport::MsgReportChannelSampleRateChanged: 1000000 S/s
[... cut ...]
2022-06-05 14:58:16.375 (D) USRPInputThread::issueStreamCmd  stop
2022-06-05 14:58:16.375 (D) USRPInputThread::stopWork: stream stopped
2022-06-05 14:58:16.375 (D) USRPInput::suspendRxBuddies (0)
2022-06-05 14:58:16.375 (D) USRPInput::suspendTxBuddies (0)
2022-06-05 14:58:16.376 (D) USRPInput::resumeTxBuddies (0)
2022-06-05 14:58:16.376 (D) USRPInput::resumeRxBuddies (0)
2022-06-05 14:58:16.376 (D) GLSpectrumGUI::handleInputMessages: message: SpectrumVis::MsgStartStop
2022-06-05 14:58:16.376 (D) BFMDemod::stop

Bob-Sellers, 2022-06-06:
I think there is an issue with the setting of the sample rate.

I select 1MS/s and tune to a prominent signal in the Broadcast FM band. I then increase the frequency in small steps and track by eye the prominent signal. I would expect to be able to tune close to 0.5MHz high before the prominent signal disappears off the left hand side of the spectrum. However, I can still see the signal when tuned 1.9 MHz higher. This is of course symettrical, so I can still see the signal when I'm tuned 1.9 MHz low.

If I reduce the sample rate, using the thumbwheel digits next to the 'SR' label to 250,000 S/s , then my prominent signal is at the edges of the spectrum when the center frequency is +/- 0.5 MHz.

From what I've seen, it looks like the radio is sampling 4x faster than it is asked to.

srcejon, 2022-06-06:
Possibly something works differently on the E320 compared to B210, with regard to master clock rate.

Look for these messages in the log file and copy and paste what you get:

"USRPInput::applySettings: actual sample rate ...
"USRPInput::applySettings: master_clock_rate ...

Bob-Sellers, 2022-06-06:
I set sample rate in the GUI at 1MS/s and here are the log entries:

2022-06-06 11:48:08.661 (D) USRPInput::applySettings: clock set to  "internal"
2022-06-06 11:48:08.661 (D) USRPInput::applySettings: set sample rate set to 1000000
2022-06-06 11:48:08.771 (D) USRPInput::applySettings: frequency set to 93000000 with LO offset 0
2022-06-06 11:48:08.803 (D) USRPInput::applySettings: AGC enabled for channel  0
2022-06-06 11:48:08.804 (D) USRPInput::applySettings: set antenna path to TX/RX on channel 0
2022-06-06 11:48:08.809 (D) USRPInput::applySettings: actual sample rate 1000000.000000
2022-06-06 11:48:08.809 (D) USRPInput::applySettings: master_clock_rate 61440000.000000
2022-06-06 11:48:08.809 (D) USRPInput::applySettings: forward change to all buddies
2022-06-06 11:48:08.811 (D) USRPInput::applySettings: clock source is  "internal"
2022-06-06 11:48:08.811 (D) USRPInput::applySettings: center freq:  93000000  Hz  m_transverterMode:  false  m_transverterDeltaFrequency:  0  deviceCenterFrequency:  93000000  device stream sample rate:  1,000,000 S/s  sample rate with soft decimation:  1,000,000 S/s  m_log2SoftDecim:  0  m_gain:  50  m_lpfBW:  10,000,000  m_antennaPath:  TX/RX  m_clockSource:  internal  force:  true

srcejon, 2022-06-06:
With the b210, I get a different master_clock_rate:

2022-06-06 11:59:11.879 (D) USRPInput::applySettings: set sample rate set to 1000000
2022-06-06 11:59:11.880 (D) USRPInput::applySettings: actual sample rate 1000000.000000
2022-06-06 11:59:11.880 (D) USRPInput::applySettings: master_clock_rate 32000000.000000

But the "actual sample rate" is what the UHD driver thinks it has been set too, which looks correct.

srcejon, 2022-06-06:
I think the problem is related to your master_clock_rate being 61440000.

According to https://kb.ettus.com/E320_Getting_Started_Guide,  this should be an integer multiple of the sample rate (which that isn't).

Bob-Sellers, 2022-06-06:
Was just looking at the same doc. It does say that the default master clock for E320 is 16 MHz, but it can go up to 61.44 MHz. 

In https://files.ettus.com/manual/page_general.html#general_sampleratenotes It does say that if the master_clock_rate is not specified, then the UHD driver (3.8.5 and newer)  will automatically select a clock rate that will work with the requested sample rate. Looks like this might be the better solution for B2XX and E3XX radios.

Could I set a command line argument to set the master clock rate or does this parameter get overwritten every time a capture is started?

I'll have a play.

srcejon, 2022-06-06:
SDRangel doesn't set the master clock, I believe - that's just displaying what UHD has set.

You can try adding master_clock_rate=xx  after ip_addr parameter, separated by ,

E.g. something like

ip_addr=192.168.0.1,master_clock_rate=16000000

Bob-Sellers, 2022-06-06:
I've tried adding the 'master_clock_rate' parameter as a 'Device user argument' but as yet have not had any success. I'll keep trying.

Incidentally, I used to code in the ip_addr as a user parameter but have found that I don't need it. I deleted all 'Device user arguments' for my USRP device and now I can select the RF A or RF B channel on the E320 by selecting either sampling device 'ni-e320-XXXXXXX[0:0]' or 'ni-e320-XXXXXXX[0:1]'

I wonder whether knowledge of the ip address has persisted - maybe an uninstall and reinstall would test this?

Bob-Sellers, 2022-06-06:
If I use the Ettus supplied command line utility to capture samples, with the parameters I've mentioned above I get warnings saying the sample rate can't be achieved. eg **stipulating master clock rate with this:**

rx_samples_to_file --args "master_clock_rate = 61440000" --file c:\temp\usrp_samples.dat --rate 1000000 --freq 93000000

I get this:

Setting RX Rate: 1.000000 Msps...
[WARNING] [0/DDC#0] The requested decimation is odd; the user should expect passband CIC rolloff.
Select an even decimation to ensure that a halfband filter is enabled.
Decimations factorable by 4 will enable 2 halfbands, those factorable by 8 will enable 3 halfbands.
decimation = dsp_rate/samp_rate -> 61
Actual RX Rate: [WARNING] [0/DDC#0] The requested decimation is odd; the user should expect passband CIC rolloff.
Select an even decimation to ensure that a halfband filter is enabled.
Decimations factorable by 4 will enable 2 halfbands, those factorable by 8 will enable 3 halfbands.
decimation = dsp_rate/samp_rate -> 61
1.007213 Msps...[WARNING] [MULTI_USRP] Could not set RX rate to 1.000 MHz. Actual rate is 1.007 MHz

**If I don't stipulate the master_clock_rate ie I use this:**

rx_samples_to_file --file c:\temp\usrp_samples.dat --rate 1000000 --freq 93000000

I get this:

Setting RX Rate: 1.000000 Msps...
Actual RX Rate: 1.000000 Msps...

Setting RX Freq: 93.000000 MHz...
Setting RX LO Offset: 0.000000 MHz...
Actual RX Freq: 93.000000 MHz...

Waiting for "lo_locked": ++++++ locked.

SDRangel seems to be reporting that it is using a master_clock_rate and sample rate which are not possible?

srcejon, 2022-06-06:
> SDRangel seems to be reporting that it is using a master_clock_rate and sample rate which are not possible?

Yes - as I said, the master clock needs to be an integer multiple of the sample rate.

I suspect the problem is that at startup, to work out range of supported sample rates, SDRangel sets master clock to min and max values. It then tries to reset it to automatic, with:

    properties->access<bool>("/mboards/0/auto_tick_rate").set(true);

I'm guessing that this isn't the way to do it on the E320, as it appears you have the max clock rate still set (61440000).

Perhaps the property name is different. Can you send me the output of:

    uhd_usrp_probe --tree

This will list all the properties for your device.

Bob-Sellers, 2022-06-06:
Here's 'uhd_usrp_probe --tree' without the stuff at the top

/
/name
/mboards
/mboards/0
/mboards/0/name
/mboards/0/serial
/mboards/0/connection
/mboards/0/link_max_rate
/mboards/0/mpm_version
/mboards/0/fpga_version
/mboards/0/fpga_version_hash
/mboards/0/token
/mboards/0/args
/mboards/0/mender_artifact
/mboards/0/mpm_sw_version
/mboards/0/fs_version
/mboards/0/clock_source
/mboards/0/clock_source/value
/mboards/0/clock_source/options
/mboards/0/time_source
/mboards/0/time_source/value
/mboards/0/time_source/options
/mboards/0/sensors
/mboards/0/sensors/ref_locked
/mboards/0/sensors/gps_locked
/mboards/0/sensors/fan
/mboards/0/sensors/temp_fpga
/mboards/0/sensors/temp_internal
/mboards/0/sensors/temp_rf_channelA
/mboards/0/sensors/temp_rf_channelB
/mboards/0/sensors/temp_main_power
/mboards/0/sensors/gps_gpgga
/mboards/0/sensors/gps_sky
/mboards/0/sensors/gps_time
/mboards/0/sensors/gps_tpv
/mboards/0/eeprom
/mboards/0/components
/mboards/0/components/fpga
/mboards/0/components/dts
/blocks
/blocks/0
/blocks/0/DUC#0
/blocks/0/DUC#0/noc_id
/blocks/0/DDC#0
/blocks/0/DDC#0/noc_id
/blocks/0/Radio#0
/blocks/0/Radio#0/noc_id
/blocks/0/Radio#0/dboard
/blocks/0/Radio#0/dboard/tx_frontends
/blocks/0/Radio#0/dboard/tx_frontends/0
/blocks/0/Radio#0/dboard/tx_frontends/0/name
/blocks/0/Radio#0/dboard/tx_frontends/0/connection
/blocks/0/Radio#0/dboard/tx_frontends/0/antenna
/blocks/0/Radio#0/dboard/tx_frontends/0/antenna/value
/blocks/0/Radio#0/dboard/tx_frontends/0/antenna/options
/blocks/0/Radio#0/dboard/tx_frontends/0/freq
/blocks/0/Radio#0/dboard/tx_frontends/0/freq/value
/blocks/0/Radio#0/dboard/tx_frontends/0/freq/range
/blocks/0/Radio#0/dboard/tx_frontends/0/bandwidth
/blocks/0/Radio#0/dboard/tx_frontends/0/bandwidth/value
/blocks/0/Radio#0/dboard/tx_frontends/0/bandwidth/range
/blocks/0/Radio#0/dboard/tx_frontends/0/gains
/blocks/0/Radio#0/dboard/tx_frontends/0/gains/PGA
/blocks/0/Radio#0/dboard/tx_frontends/0/gains/PGA/value
/blocks/0/Radio#0/dboard/tx_frontends/0/gains/PGA/range
/blocks/0/Radio#0/dboard/tx_frontends/1
/blocks/0/Radio#0/dboard/tx_frontends/1/name
/blocks/0/Radio#0/dboard/tx_frontends/1/connection
/blocks/0/Radio#0/dboard/tx_frontends/1/antenna
/blocks/0/Radio#0/dboard/tx_frontends/1/antenna/value
/blocks/0/Radio#0/dboard/tx_frontends/1/antenna/options
/blocks/0/Radio#0/dboard/tx_frontends/1/freq
/blocks/0/Radio#0/dboard/tx_frontends/1/freq/value
/blocks/0/Radio#0/dboard/tx_frontends/1/freq/range
/blocks/0/Radio#0/dboard/tx_frontends/1/bandwidth
/blocks/0/Radio#0/dboard/tx_frontends/1/bandwidth/value
/blocks/0/Radio#0/dboard/tx_frontends/1/bandwidth/range
/blocks/0/Radio#0/dboard/tx_frontends/1/gains
/blocks/0/Radio#0/dboard/tx_frontends/1/gains/PGA
/blocks/0/Radio#0/dboard/tx_frontends/1/gains/PGA/value
/blocks/0/Radio#0/dboard/tx_frontends/1/gains/PGA/range
/blocks/0/Radio#0/dboard/rx_frontends
/blocks/0/Radio#0/dboard/rx_frontends/0
/blocks/0/Radio#0/dboard/rx_frontends/0/name
/blocks/0/Radio#0/dboard/rx_frontends/0/connection
/blocks/0/Radio#0/dboard/rx_frontends/0/antenna
/blocks/0/Radio#0/dboard/rx_frontends/0/antenna/value
/blocks/0/Radio#0/dboard/rx_frontends/0/antenna/options
/blocks/0/Radio#0/dboard/rx_frontends/0/freq
/blocks/0/Radio#0/dboard/rx_frontends/0/freq/value
/blocks/0/Radio#0/dboard/rx_frontends/0/freq/range
/blocks/0/Radio#0/dboard/rx_frontends/0/bandwidth
/blocks/0/Radio#0/dboard/rx_frontends/0/bandwidth/value
/blocks/0/Radio#0/dboard/rx_frontends/0/bandwidth/range
/blocks/0/Radio#0/dboard/rx_frontends/0/gains
/blocks/0/Radio#0/dboard/rx_frontends/0/gains/PGA
/blocks/0/Radio#0/dboard/rx_frontends/0/gains/PGA/value
/blocks/0/Radio#0/dboard/rx_frontends/0/gains/PGA/range
/blocks/0/Radio#0/dboard/rx_frontends/1
/blocks/0/Radio#0/dboard/rx_frontends/1/name
/blocks/0/Radio#0/dboard/rx_frontends/1/connection
/blocks/0/Radio#0/dboard/rx_frontends/1/antenna
/blocks/0/Radio#0/dboard/rx_frontends/1/antenna/value
/blocks/0/Radio#0/dboard/rx_frontends/1/antenna/options
/blocks/0/Radio#0/dboard/rx_frontends/1/freq
/blocks/0/Radio#0/dboard/rx_frontends/1/freq/value
/blocks/0/Radio#0/dboard/rx_frontends/1/freq/range
/blocks/0/Radio#0/dboard/rx_frontends/1/bandwidth
/blocks/0/Radio#0/dboard/rx_frontends/1/bandwidth/value
/blocks/0/Radio#0/dboard/rx_frontends/1/bandwidth/range
/blocks/0/Radio#0/dboard/rx_frontends/1/gains
/blocks/0/Radio#0/dboard/rx_frontends/1/gains/PGA
/blocks/0/Radio#0/dboard/rx_frontends/1/gains/PGA/value
/blocks/0/Radio#0/dboard/rx_frontends/1/gains/PGA/range
/blocks/0/Radio#0/rx_frontends
/blocks/0/Radio#0/rx_frontends/0
/blocks/0/Radio#0/rx_frontends/0/sensors
/blocks/0/Radio#0/rx_frontends/0/sensors/ad9361_temperature
/blocks/0/Radio#0/rx_frontends/0/sensors/rssi
/blocks/0/Radio#0/rx_frontends/0/sensors/lo_lock
/blocks/0/Radio#0/rx_frontends/0/sensors/lo_locked
/blocks/0/Radio#0/rx_frontends/1
/blocks/0/Radio#0/rx_frontends/1/sensors
/blocks/0/Radio#0/rx_frontends/1/sensors/ad9361_temperature
/blocks/0/Radio#0/rx_frontends/1/sensors/rssi
/blocks/0/Radio#0/rx_frontends/1/sensors/lo_lock
/blocks/0/Radio#0/rx_frontends/1/sensors/lo_locked
/blocks/0/Radio#0/tx_frontends
/blocks/0/Radio#0/tx_frontends/0
/blocks/0/Radio#0/tx_frontends/0/sensors
/blocks/0/Radio#0/tx_frontends/0/sensors/ad9361_temperature
/blocks/0/Radio#0/tx_frontends/0/sensors/lo_lock
/blocks/0/Radio#0/tx_frontends/0/sensors/lo_locked
/blocks/0/Radio#0/tx_frontends/1
/blocks/0/Radio#0/tx_frontends/1/sensors
/blocks/0/Radio#0/tx_frontends/1/sensors/ad9361_temperature
/blocks/0/Radio#0/tx_frontends/1/sensors/lo_lock
/blocks/0/Radio#0/tx_frontends/1/sensors/lo_locked
/blocks/0/Radio#0/rx_codec
/blocks/0/Radio#0/rx_codec/name
/blocks/0/Radio#0/tx_codec
/blocks/0/Radio#0/tx_codec/name
/blocks/0/DmaFIFO#0
/blocks/0/DmaFIFO#0/noc_id

Bob-Sellers, 2022-06-06:
From the log below, it looks like the correct sample rate is achieved using the auto setting of the master_clock_rate via UHD (at 16:53:13.671), and then the master_clock_rate is manually configured, which scuppers it?

The actual sample rate isn't reported after the master_clock_rate is set, I wonder if it changes?

2022-06-06 16:53:13.519 (D) USRPInput::applySettings: clock set to  "internal"
2022-06-06 16:53:13.519 (D) USRPInput::applySettings: set sample rate set to 1000000
2022-06-06 16:53:13.632 (D) USRPInput::applySettings: frequency set to 93000000 with LO offset 0
2022-06-06 16:53:13.665 (D) USRPInput::applySettings: AGC enabled for channel  0
2022-06-06 16:53:13.665 (D) USRPInput::applySettings: set antenna path to TX/RX on channel 0
2022-06-06 16:53:13.671 (D) USRPInput::applySettings: actual sample rate 1000000.000000
2022-06-06 16:53:13.671 (D) USRPInput::applySettings: master_clock_rate 61440000.000000
2022-06-06 16:53:13.671 (D) USRPInput::applySettings: forward change to all buddies
2022-06-06 16:53:13.673 (D) USRPInput::applySettings: clock source is  "internal"

srcejon, 2022-06-07:
According to the USRP mailing list, E320 doesn't support automatic master clock (despite what it says in the docs).

So the above patch skips the min/max sample rate calculation, that is now just for B2xx. The master clock is just left as the default. You may be able to adjust it with the device arg, if needed.

Can you give this build a try? https://github.com/f4exb/sdrangel/suites/6821039503/artifacts/262569799

Bob-Sellers, 2022-06-07:
Just given this a quick go.

master_clock_rate appears to be set to 16 MHz as indicated in the sampling device pane.

setting sample rate to 1MS/s looks like it works correctly, the LO can be swung +/- 0.5MHz and a known signal stays in the spectrum view.

Broadcast FM demodulator works well, where it previously struggled (dropping audio).

ADSB is picking up some aircraft.

all looks good, will play around with it a bit more, but thought i'd just give you a bit of brief feedback.

Excellent work and thank you

Bob-Sellers, 2022-06-07:
Extract from log with sample rate set to 2MS/s:

2022-06-07 17:51:21.090 (D) USRPInput::applySettings: clock set to  "internal"
2022-06-07 17:51:21.091 (D) USRPInput::applySettings: set sample rate set to 2000000
2022-06-07 17:51:21.203 (D) USRPInput::applySettings: frequency set to 93000000 with LO offset 0
2022-06-07 17:51:21.244 (D) USRPInput::applySettings: AGC enabled for channel  0
2022-06-07 17:51:21.244 (D) USRPInput::applySettings: set antenna path to TX/RX on channel 0
2022-06-07 17:51:21.251 (D) USRPInput::applySettings: actual sample rate 2000000.000000
2022-06-07 17:51:21.251 (D) USRPInput::applySettings: master_clock_rate 16000000.000000

srcejon, 2022-06-07:
Good to hear.

What happens if you try to set a sample rate that isn't an integer factor of the master clock? Does it go wrong or just correct to the nearest valid value?

srcejon, 2022-06-07:
Good, thanks.

The master rate arg wouldn't have worked properly before, but hopefully it does now.

Bob-Sellers, 2022-06-07:
One thing I have noticed is that when I occasionally get a 'timeout' I cannot clear the error / restart the sampling device by clicking on the stop and then start button. I need to 'Reload device'.

Having said that I've just had an occurrence when I can't clear the timeout error via a reload

..... or deleting and re adding the sampling device

in fact I had to close SDRangel and restart it.

The timeout normally happens when I set too high a sample rate.

Bob-Sellers, 2022-06-07:
Just had some success setting the master_clock_rate via the device arguments.

Need to 'Reload Device' for changes to take effect but then new sample rates that were previously unavailable now are (presumably within the limitations of the E320 interpolators/decimators).

setting in excess of 61.44MHz results in 'could not start sample source' error, as expected.

srcejon, 2022-06-15:
Fixed in 7.3.2.