f4exb/sdrangel #1279: HackRF transmits okay first time, stops on second time, and okay third time, and repeats (add even). Why?

electrical-pro, 2022-06-05:
https://user-images.githubusercontent.com/31592485/172065502-c15ec659-4b09-4887-8843-92f0582e70f2.mp4

Weird issue with HackRF One (please watch the video).
When I press the start/stop button HackRF transmits on add times but not on even.
On odd times everything is fine.
On even times, there is a carrier, but no modulation and the carrier seems to be weaker.
I rarely use Hack RF for transmitting so I don't know when this issue started, maybe it's related to the firmware update (release 2021.03.1), but I'm not sure.

here are the settings I used
[image]
sdrangel-7.3.1

I also tested with old sdrangel-4.14.3 and it is the same.

I used sdrangel-4.14.3 a few years ago without any issues, so I assume it may be related to HackRF firmware update (release 2021.03.1).

What is causing this issue?

f4exb, 2022-06-06:
This is normal. HackRF has a residual carrier remaining at the last frequency it was set.

electrical-pro, 2022-06-06:
@f4exb, I still do not fully understand. Why there is no modulation?
Maybe I explained the issue poorly or I don't understand something.
I expect HackRF to transmit a modulated signal every time it goes into TX mode.
But it only transmits modulated TX signal on odd times but not on even times.

Expected behaviour:
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF

Observed behaviour:
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF
start/stop button clicked - transmission WITHOUT modulation is ON (why?)
start/stop button clicked - transmission WITHOUT modulation is OFF
start/stop button clicked - transmission with modulation is ON
start/stop button clicked - transmission with modulation is OFF
start/stop button clicked - transmission WITHOUT modulation is ON
start/stop button clicked - transmission WITHOUT modulation is OFF

f4exb, 2022-06-06:
For me it works normally starting with modulation at each ON.

electrical-pro, 2022-06-06:
@f4exb, may I ask if you using the latest HackRF firmware release 2021.03.1?

[image]

I tested on another computer with the same issue. So I assume it's somehow related to my HackRF, it doesn't look like a hardware issue, so I think the latest firmware causes this.

Can anyone confirm the issue on HackRF firmware release 2021.03.1?

f4exb, 2022-06-06:
It is much older than that:
```
hackrf_info version: git-cc7f599                                                                                                                                                                                                            
libhackrf version: git-cc7f599 (0.5)                                                                                                                                                                                                        
Found HackRF                                                                                                                                                                                                                                
Index: 0                                                                                                                                                                                                                                    
Serial number: 0000000000000000088869dc3878a51b                                                                                                                                                                                             
Board ID Number: 2 (HackRF One)                                                                                                                                                                                                             
Firmware Version: 2017.02.1 (API:1.02)                                                                                                                                                                                                      
Part ID Number: 0xa000cb3c 0x006a4761   
```

Radiomix2000, 2022-06-06:
> @f4exb, may I ask if you using the latest HackRF firmware release 2021.03.1?

Did you update CPLD too?

closed 2022-08-12 by github-actions[bot]

electrical-pro, 2022-10-17:
Fixed the issue by updating to [release 2022.09.1](https://github.com/greatscottgadgets/hackrf/releases/tag/v2022.09.1)