f4exb/sdrangel #1284: USRP X310 - failed to open device

Bob-Sellers, 2022-06-10:
Windows 10, with SDRangel 7.3.1-19-g6029d3089

Ettus X310 fitted with 2 x TwinRX daughterboards (ie radio has 4 RX channels and 0 TX channels)

I was wanting to perform some Noise Figure measurements on an Ettus X310 radio using SDRangel's excellent new support for networked USRP radios. However, I've come across a problem. I can't 'open' the radio.

The log reads as follows:

2022-06-10 14:18:44.243 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2022-06-10 14:18:44.244 (D) MainWindow::sampleSourceCreate: deviceSetIndex: 0 deviceIndex: 6 hardwareId: "USRP" sequence: 0 id: "sdrangel.samplesource.usrp" serial: "type=x300,addr=192.168.10.2,fpga=HG,name=,serial=XXXXXXX,product=X310" displayedName: "[0:0] XXXXXXX"
2022-06-10 14:18:44.244 (D) USRPInput::suspendRxBuddies (0)
2022-06-10 14:18:44.244 (D) USRPInput::suspendTxBuddies (0)
2022-06-10 14:18:44.249 (D) USRPInput::openDevice: allocated SampleFifo
2022-06-10 14:18:44.249 (D) USRPInput::openDevice: open device here
2022-06-10 14:18:44.249 (D) DeviceUSRPParams::open: type=x300,addr=192.168.10.2,fpga=HG,name=,serial=XXXXXXX,product=X310
2022-06-10 14:18:47.484 (D) DeviceUSRPParams::open: exception:  LookupError: KeyError: Invalid TX channel: 0
2022-06-10 14:18:47.484 (C) USRPInput::openDevice: failed to open device
2022-06-10 14:18:47.484 (D) USRPInput::resumeTxBuddies (0)
2022-06-10 14:18:47.484 (D) USRPInput::resumeRxBuddies (0)
2022-06-10 14:18:47.491 (D) DSPDeviceSourceEngine::setSource
2022-06-10 14:18:47.492 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPSetSource
2022-06-10 14:18:47.492 (D) DSPDeviceSourceEngine::gotoIdle
2022-06-10 14:18:47.492 (D) DSPDeviceSourceEngine::handleSetSource: set USRPInput
2022-06-10 14:18:47.495 (D) DeviceGUI::DeviceGUI: 0x0
2022-06-10 14:18:47.504 (D) USRPInput::getLORange: exception:  ValueError: meta-range cannot be empty
2022-06-10 14:18:47.504 (D) USRPInput::getSRRange: exception:  ValueError: meta-range cannot be empty
2022-06-10 14:18:47.504 (D) USRPInput::getGainRange: exception:  ValueError: meta-range cannot be empty
2022-06-10 14:18:47.504 (D) USRPInput::getLORange: exception:  ValueError: meta-range cannot be empty
2022-06-10 14:18:47.504 (D) USRPInputGUI::updateFrequencyLimits: delta: 0 min: 0 max: 0
2022-06-10 14:18:47.504 (D) USRPInput::getSRRange: exception:  ValueError: meta-range cannot be empty
2022-06-10 14:18:47.505 (C) USRPInput::applySettings: could not set clock  "internal"
2022-06-10 14:18:47.505 (D) USRPInput::applySettings: forward change to all buddies

As stated at the top, the radio has four receive paths and no transmit paths so I'm wondering whether that is implicated?

srcejon, 2022-06-13:
Yes, from the exception: LookupError: KeyError: Invalid TX channel: 0, it's very likely because it has no transmit path.

The question is which method causes the exception. It may be a bit of trial-and-error to find a fix.

Bob-Sellers, 2022-06-13:
Thanks for picking this up.

I've got the radio at the ready so can do some testing when required.

PR #1288 (2022-06-13): USRP: Don't call get_tx/rx methods if tx/rx channels is 0, as may result in KeyError exception
First attempt to fix #1284
closed 2022-06-13 by f4exb