f4exb/sdrangel #1286: Button does not work with the "radio clock" channel

choupette-change, 2022-06-10:
Hello, "common settings" button does not work with the "radio clock" channel (perhaps with other untested channels). OK with "AM demodulator"
[image]

f4exb, 2022-06-11:
Probably the signal connection is missing

f4exb, 2022-06-11:
Also on a different note the menu does not appear at the right position for features because the signal emission is  coded like this:

```c++
void FeatureGUI::activateSettingsDialog()
{
    QPoint p = mapFromGlobal(QCursor::pos());
    m_contextMenuType = ContextMenuChannelSettings;
    emit customContextMenuRequested(p);
}
```
While it should be like this:

```c++
void DeviceGUI::activateSettingsDialog()
{
    QPoint p = QCursor::pos();
    m_contextMenuType = ContextMenuDeviceSettings;
    emit customContextMenuRequested(p);
}
```
Notice the difference with `QPoint`

commit 7eeb9fd7 (2022-06-12): Radio Clock: fixed common channel settings menu trigger. Fixes #1286
closed 2022-06-12 by f4exb