f4exb/sdrangel #1287: Crashes when my position is changed

doftheworld, 2022-06-12:
I am finding that any time I attempt to change "my position" under "preferences" on Mac OS version 7.3.1 the application crashes as soon as I click "ok." I had the same issue in 7.1.0 and updated to 7.3.1, but that did not resolve the issue.

srcejon, 2022-06-12:
Seems OK for me.

What features / channels do you have open?

What position do you enter (does it matter)?

Screenshot pre-crash might help.

doftheworld, 2022-06-12:
To make things clean, I closed now the only channel I had open (ads-b demodulator). After closing it, the problem was resolved. So, it seems that changing your position while having the ads-b demodulator open on Mac causes the crash. I was able to do the same procedure on Windows without issue.  

So in a round about way you solved the problem for me - so thanks! But it is perhaps a problem - since it seems you should be able to change your position while the ads-b demodulator channel is open.

srcejon, 2022-06-13:
Doesn't seem to be a problem on Windows.

loewal, 2022-06-13:
Launched the DMG-version: crashed!
Launched again: doesn't startup anymore.

Launched my selfcompiled version: OK!!

hambyte, 2022-06-24:
Same issue on 7.3.2-win64 on Windows 10. I have also been using the ADS-B feature during the times I've tried to change "My Position" and after clicking "ok" in that dialog window.

srcejon, 2022-09-20:
Crash occurs when airport database doesn't exist. Should be fixed in above patch.

closed 2022-09-27 by srcejon