f4exb/sdrangel #1290: Compiler error

SA7BNT, 2022-06-13:
Hi,

running into this compiler error 
[image]
I already tried to reinstall SDL2 but it didn't make the thing. 
How can i fix this error?

OSx is Big Sur (11.6.6)

srcejon, 2022-06-14:
I compile it on a Mac that doesn't have SDL2 installed. 

You could try changing the compilation options listed so implicit function defs aren't treated as errors.

loewal, 2022-06-14:
Compiling went OK.

But I also always install the DMG when it is available.
That last one didn’t startup…until this morning:

I ran my own compiled version and after that the DMG (not at the same time).

Now it’s ok.

Sometimes………………………….??

loewal, 2022-06-14:
By the way….SDL2?

Never installed it and until v7.3.1 all went fine…until yesterday.
A good nights sleep…trying v7.3.2 again….yes!!

So….puzzles….

closed 2022-08-19 by github-actions[bot]