f4exb/sdrangel #1295: GLScope::cleanup can be called afger GLScope destructor is called

srcejon, 2022-06-20:
It seems GLScope::cleanup can be called after the GLScope destructor is called, due to this connect:

    connect(glCurrentContext, &QOpenGLContext::aboutToBeDestroyed, this, &GLScope::cleanup);

There's also a similar problem in GLSpectrum.

It can be observed by putting qDebug in the destructor and cleanup methods and then closing the Channel Analyser.

It doesn't seem to cause a crash for me - but potentially could. I seem to recall someone mentioned a crash when the Channel Analyser was closed, but wasn't always reproducible. Can't find the Issue Id, but quite possibly related to this.

srcejon, 2022-06-21:
Do you not have to call cleanup even if glCurrentContext is null? Otherwise there will be objects in the glshader classes that are not deleted and thus a small memory leak.

Perhaps cleanup needs to be in two parts - things like glDeleteTextures that should only happen if there is a context, and the deletes that should happen regardless?

f4exb, 2022-06-21:
`glCurrentContext` is not the current context but a private reference to the current context in the `GLScope` class that gets initialized in the 'initializeGL` method thus checking for null is like checking we went through  'initializeGL` and initialized the shaders.

However I am wondering how this can work because the various shadrers are memebers of `GLScope` and thus their destructor will be called when `GLScope` is destroyed and in the destructor code of the shader its `cleanup` method is called. Logically this could mean that the `cleanup` method of the shaders are called several time and they don't seem to be protected against this situation so I am wondering why there are not many more issues. 

Maybe it would be better to let each shader take care of its own cleanup in its destructor.

srcejon, 2022-06-21:
> Logically this could mean that the `cleanup` method of the shaders are called several times and they don't seem to be protected against this situation

The ones I have a quick look at seem OK. They set the pointers to nullptr - and it's OK to call delete with a nullptr if called again. Likewise, textureIds are reset to 0, so glDeleteTextures shouldn't be called twice.

closed 2022-06-28 by f4exb