f4exb/sdrangel #1296: Macos compiler error

loewal, 2022-06-20:
This one:
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework
CMake Error at /usr/local/opt/qt@5/lib/cmake/Qt5WebEngineCore/Qt5WebEngineCoreConfig.cmake:93 (find_package):
  Could not find a configuration file for package "Qt5WebChannel" that is
  compatible with requested version "5.15.10".

  The following configuration files were considered but not accepted:

    /usr/local/opt/qt@5/lib/cmake/Qt5WebChannel/Qt5WebChannelConfig.cmake, version: 5.15.5

Call Stack (most recent call first):
  /usr/local/opt/qt@5/lib/cmake/Qt5WebEngine/Qt5WebEngineConfig.cmake:93 (find_package)
  /usr/local/opt/qt@5/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  CMakeLists.txt:345 (find_package)

Are there additional dependencies now that 3D is implemented?

loewal, 2022-06-20:
I could compile it on my Macbook after I install qt5-qtwebchannel with Macports.
My iMac is until now configured through Macports...
I will try 7.3.2

loewal, 2022-06-20:
You're right Jon,
Same error...could you please help?
I already did brew reinstall qt5

srcejon, 2022-06-20:
I don't know. But sounds like you have a mismatch between two different version. 5.15.10 and 5.15.5.

loewal, 2022-06-22:
Hi Jon,
After I made some corrections everything works again. It's only strange what has happened.

I changed the test in:
/usr/local/opt/qt@5/lib/cmake/Qt5WebEngineWidgets/Qt5WebEngineWidgetsConfig.cmake
and 
/usr/local/opt/qt@5/lib/cmake/Qt5WebEngineCore/Qt5WebEngineCoreConfig.cmake

changed lines like set(Qt5WebEngineWidgets_VERSION_STRING 5.15.10) to 5.15.5 and that worked (there were some more...).
I installed the latest version with Homebrew, so......?

srcejon, 2022-06-22:
Ok - I'd say that is a qt / Homebrew issue, rather than SDRangel - so probably best taken up Homebrew.

closed 2022-06-22 by srcejon